carla-simulator/carla #7228: Moving Vehicle Along Specified Route

mariashaima, 2024-03-08:
Problem :I want to move my vehicle along this route: [124, 30, 31, 33, 84, 36, 32, 23, 128, 70, 129, 28]. What should I do?
my current code is :

# Get route indices
        route_1_indices = [124, 30, 31, 33, 84, 36, 32, 23, 128, 70, 129, 28]
       
        spawn_point = world.get_map().get_spawn_points()[124]
        # Spawn the vehicle at the specified spawn point
        vehicle = world.spawn_actor(blueprint, spawn_point)
        
        # Set autopilot to true to make the vehicle move automatically
        vehicle.set_autopilot(True)
        
        # Draw debug strings for waypoints
        for ind in route_1_indices:
            world.debug.draw_string(world.get_map().get_spawn_points()[ind].location, str(ind), life_time=3600, color=carla.Color(255, 0, 0))
            
        
        actor_list.append(vehicle)

siddharthjain38, 2024-03-20:
@mariashaima refer [Traffic manager
](https://carla.readthedocs.io/en/latest/tuto_G_traffic_manager/#specify-routes-for-vehicles)

`spawn_points = world.get_map().get_spawn_points()

# Route 1
spawn_point_1 =  spawn_points[32]
# Create route 1 from the chosen spawn points
route_1_indices = [129, 28, 124, 33, 97, 119, 58, 154, 147]
route_1 = []
for ind in route_1_indices:
    route_1.append(spawn_points[ind].location)

# Route 2
spawn_point_2 =  spawn_points[149]
# Create route 2 from the chosen spawn points
route_2_indices = [21, 76, 38, 34, 90, 3]
route_2 = []
for ind in route_2_indices:
    route_2.append(spawn_points[ind].location)

# Now let's print them in the map so we can see our routes
world.debug.draw_string(spawn_point_1.location, 'Spawn point 1', life_time=30, color=carla.Color(255,0,0))
world.debug.draw_string(spawn_point_2.location, 'Spawn point 2', life_time=30, color=carla.Color(0,0,255))

for ind in route_1_indices:
    spawn_points[ind].location
    world.debug.draw_string(spawn_points[ind].location, str(ind), life_time=60, color=carla.Color(255,0,0))

for ind in route_2_indices:
    spawn_points[ind].location
    world.debug.draw_string(spawn_points[ind].location, str(ind), life_time=60, color=carla.Color(0,0,255))

while True:
    world.tick()`

davidscientistcom, 2024-04-08:
#################################################################################################################
#  David Granados Zafra.                                                                                        #
#  En este ejercicio vamos a aprender rutas
#################################################################################################################
import random
import time

import carla
from agents.navigation.global_route_planner import GlobalRoutePlanner

def init_map():
    # connect to the sim
    client = carla.Client('localhost', 2000)
    client.set_timeout(10.0)

    # optional to load different towns
    client.load_world('Town02')

    # define environment/world and get possible places to spawn a car
    world = client.get_world()

    location = carla.Location(x=100, y=200, z=300)  # Ajusta las coordenadas x, y, z según sea necesario
    rotation = carla.Rotation(pitch=-90, yaw=0, roll=0)

    # Crea la transformación con la ubicación y rotación
    transform = carla.Transform(location, rotation)
    spectator = world.get_spectator()
    spectator.set_transform(transform)
    return world, client

def add_vehicle_to_world(world, start_point, vehicles, name):
    vehicle_bp = world.get_blueprint_library().filter(name)[0]
    vehicle = world.try_spawn_actor(vehicle_bp, start_point)
    vehicles.append(vehicle)
    return vehicle

if __name__ == '__main__':
    vehicles = list()
    # Incializamos el mapa para la simulación
    world, client = init_map()

    spawn_points = world.get_map().get_spawn_points()
    start_point = spawn_points[42]
    end_point = spawn_points[17]
    # Añadimos los vehículos en las pociones correctas

    vehicle1 = add_vehicle_to_world(world, start_point, vehicles, 'vehicle.dodge.charger_police_2020')

    # Comenzamos a crear la ruta
    sampling_resolution = 1
    grp = GlobalRoutePlanner(world.get_map(), sampling_resolution)

    route = grp.trace_route(start_point.location, end_point.location)

    for waypoint in route:
        world.debug.draw_string(waypoint[0].transform.location, '^', draw_shadow=False,
                                color=carla.Color(r=0, g=0, b=255), life_time=60.0,
                                persistent_lines=True)

    vehicle1.set_autopilot(True)

    while True:
        # Carla Tick
        world.tick()

mariashaima, 2024-06-07:
In the above code, the implementation is useful, but the vehicle is not moving along the route in autopilot mode. What should I do?