carla-simulator/carla #7238: lidar

shejiadeji, 2024-03-12:
When I was using a LiDAR sensor, I found that LiDAR did not work perfectly in the repalyfile.
Some cars are visible in LiDAR sensors, while others are invisible.The category of visible cars is mesh, while the category of invisible cars is vehicle.

shejiadeji, 2024-03-12:
ubuntu 20.04
carla 0.9.14

LuisPoveda, 2024-04-19:
Thanks for posting the issue, we will look at it

Blyron, 2024-05-30:
Hello, we will need more info about what vehicles are invisible

fcheng-cv, 2024-05-31:
> Hello, we will need more info about what vehicles are invisible

During replay file, actors have no point cloud due to missing physics simulation, env objs are OK

LuisPovedaCano, 2024-05-31:
This issue has been fixed, please update to with the latest dev. Feel free to reopen it if it occurs again.

closed 2024-05-31 by LuisPovedaCano