carla-simulator/carla #7240: ImportError error: undefined symbol: _ZNSt15__exception_ptr13exception_ptr10_M_releaseEv

nischalkhanal123, 2024-03-12:
CARLA version: 0.9.15
OS: Ubuntu 22.04

I followed the tutorial for building CARLA from the source. After compiling the client and server, I wanted to test the simulator using the examples. While running the script, I get the following error:

> Traceback (most recent call last):
>   File "/home/nischalkhanal/carla/PythonAPI/examples/manual_control.py", line 83, in <module>
>     import carla
>   File "/home/nischalkhanal/carla/PythonAPI/examples/carla/__init__.py", line 8, in <module>
>     from .libcarla import *
> ImportError: /home/nischalkhanal/carla/PythonAPI/examples/carla/libcarla.cpython-39-x86_64-linux-gnu.so: undefined symbol: _ZNSt15__exception_ptr13exception_ptr10_M_releaseEv

Is there a way to fix this?

Thank you,
Nischal

MarcelPiNacy, 2024-05-03:
Hi,

Do you still have this issue?

Marcel

HansYeoh, 2025-01-21:
I have found the same problem as you, have you found a solution?