carla-simulator/carla #7246: AttributeError: 'CarlaRosBridge' object has no attribute 'shutdown'

yuan-mingxi, 2024-03-13:
(carla) yuan@yuan-asus:~/carla-ros-bridge/ros-bridge$     **roslaunch carla_ros_bridge carla_ros_bridge.launch**

... logging to /home/yuan/.ros/log/4ad68a1e-e128-11ee-95b7-135e10c5b2f5/roslaunch-yuan-asus-11995.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yuan-asus:34895/

SUMMARY
========

PARAMETERS
 * /carla_ros_bridge/ego_vehicle_role_name: ["hero", "ego_veh...
 * /carla_ros_bridge/fixed_delta_seconds: 0.05
 * /carla_ros_bridge/host: localhost
 * /carla_ros_bridge/passive: False
 * /carla_ros_bridge/port: 2000
 * /carla_ros_bridge/register_all_sensors: True
 * /carla_ros_bridge/synchronous_mode: True
 * /carla_ros_bridge/synchronous_mode_wait_for_vehicle_control_command: False
 * /carla_ros_bridge/timeout: 2
 * /carla_ros_bridge/town: Town01
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True

NODES
  /
    carla_ros_bridge (carla_ros_bridge/bridge.py)

auto-starting new master
process[master]: started with pid [12018]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4ad68a1e-e128-11ee-95b7-135e10c5b2f5
process[rosout-1]: started with pid [12043]
started core service [/rosout]
process[carla_ros_bridge-2]: started with pid [12049]
/home/yuan/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py:15: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
[INFO] [1710327368.213197, 0.000000]: Trying to connect to localhost:2000
/home/yuan/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py:416: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) != LooseVersion(CarlaRosBridge.CARLA_VERSION):
_**_[ERROR] [1710327370.218684, 0.000000]: Error: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000_**_
[INFO] [1710327370.221724, 0.000000]: Shutting down...
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/core.py", line 572, in signal_shutdown
    h()
  File "/home/yuan/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_ros_bridge/src/carla_ros_bridge/bridge.py", line 349, in destroy
    self.shutdown.set()
**AttributeError: 'CarlaRosBridge' object has no attribute 'shutdown'**
================================================================================REQUIRED process [carla_ros_bridge-2] has died!
process has finished cleanly
log file: /home/yuan/.ros/log/4ad68a1e-e128-11ee-95b7-135e10c5b2f5/carla_ros_bridge-2*.log
Initiating shutdown!
================================================================================
[carla_ros_bridge-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Anhar2, 2024-03-17:
Hi, it's a bug in the code I think, it happens when ros-bridge cannot connect to the carla server before timeout like you can see in the error produced "[ERROR] [1710327370.218684, 0.000000]: Error: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000"

You can simply increase the timeout by adding this to your ros bridge command 

`'timeout:=200'`

yuan-mingxi, 2024-03-17:
> Hi, it's a bug in the code I think, it happens when ros-bridge cannot connect to the carla server before timeout like you can see in the error produced "[ERROR] [1710327370.218684, 0.000000]: Error: [...]

hi~i finally solve the problem by reinstall carla 0.9.13 and ros-bridge in envs enviroment with python 3.7(i dont know if it is the right solution ) but after that the screen become black with only Control Interface menu but  no image，so i decided to use python instead of ros-bridge

MarcelPiNacy, 2024-05-03:
Hi,

Sorry for the delay! Yeah, CARLA (0.9.15 and older) is very sensitive to the installed python environment, ideally you should only have python 3.8 or 3.7.
We're currently working on improving these kinds of problems.
Should I mark this as completed?

Marcel

yuan-mingxi, 2024-05-03:
your email has been received, thank you!邮件已收到，谢谢！

yuan-mingxi, 2024-05-05:
Sure!So Much Thanks again!