carla-simulator/carla #7262: Houdine and Digital Twin tool

NnaYelsel, 2024-03-18:
CARLA version: **0.9.15**
Platform/OS: **Windows**
Problem you have experienced: **Could only build Carla by disable Houdini and have Carla shutdown when using UW_OnRoadMainWidget** 

Hello,

Previously, I had some problem to compile Carla on my computer because I had the following problem:
[image]

I could finally compile by following the indications of this thread: https://github.com/carla-simulator/carla/issues/7012 

But, now that I'm trying to use the digital twin tool, I got UE crashing when I'm trying to use UW_OnRoadMainWidget.

Do you think that's it's related ? 

Best regards

lightlefter, 2024-04-10:
hi,I still traped in your first question.
 #7012 didn't work well, could you explain more specific?
just delete the two codeline of  carla tool?

NnaYelsel, 2024-04-11:
Hi, 
Applying #7012 worked well to compile Carla, but then I struggle to use the Digital Twin tool, and I was wondering if the fact that I couldn't make Digital Twin Tool work was related to the modification I had to do to compile Carla, i.e disable Houdini for example.

PR #7672 (2024-05-17): Aaron/fixdigitaltwins windows

#### Description

Fix Windows version of Digital Twins

Fixes #7262 #6979   

#### Where has this been tested?

  * **Platform(s):** Windows 11
  * **Python version(s):** Don't need
  * **Unreal Engine version(s):** 4.26

#### Possible Drawbacks

Big maps cannot be created in Windows
closed 2024-05-17 by Blyron