carla-simulator/carla #7265: OpenDRIVE parts(waypoints) are missing on custom map

ronyshaji, 2024-03-19:
Hello Everyone

Hope you are having a great day. I tried to create a map in RoadRunner and imported to CARLA.
But unfortunately some of the OpenDRIVE parts are missing as shown in the image.

The OpenDRIVE does not have an issue as i checked with an .xodr viewer. But somehow it is missing after importing to carla.
In some places, even a full lane waypoint is missing.

Did anyone faced this issue? Any idea to solve this issue ?

[image]

Thanks
Roxc

gouchaoer, 2024-09-18:
same problem, have you solved it?

nikolatopic131, 2024-11-02:
I have the same problem. Half of my road network and my on-ramps don't have any waypoints for some reason even though in the opendriver export preview tool they all seem fine.

zhanghao-21, 2025-12-10:
I encountered the same issue. When I created an isolated intersection in RoadRunner and imported it into CARLA, I noticed that all outgoing lanes had waypoints, but all incoming lanes were missing them.

After some investigation, I found the cause: the incoming lanes do not generate waypoints because CARLA cannot detect a connecting road—it essentially “doesn’t know where the vehicle would come from.”

To fix this, I added another intersection connected to the incoming lanes. After doing so, the incoming lanes correctly generated waypoints. Simply adding a straight road did not work (I tested this); it specifically requires adding an additional intersection.