carla-simulator/carla #7273: How do 4.27 carsim plugins work with unreal engine 4.26

DesayHiL, 2024-03-20:
How do 4.27 carsim plugins work with unreal engine 4.26? 
This problem occurs when I compile：
[image]

PatrickPromitzer, 2024-03-26:
If you use a plugin, you should use the right unreal version of the plugin.
For example, if you use UE 4.26 and you have plugins for UE 4.27, it goes in the direction of "maybe it works, maybe it doesn't" 

From the error message "built with a different engine", I guess you don't use Carla UE4?
https://carla.readthedocs.io/en/latest/build_linux/#unreal-engine

The normal Unreal engine doesn't work with Carla

JoseM98, 2024-04-12:
As @PatrickPromitzer said, it is not recommended to use plugins made for one UE version into another version. Even if it compiles, it may give issues that were not expected.

@DesayHiL, we can't help you anymore because UE plugins are not backwards compatible. Please, try to use the modified version of UE 4.26 for CARLA as they have been tested for this one. Are you ok with closing this issue?

JoseM98, 2024-04-19:
As there has been no response I close the issue. Feel free to reopen it.

closed 2024-04-19 by JoseM98