carla-simulator/carla #7282: Error 6 installing while building Carla, make launch

evoevo07, 2024-03-22:
Hello guys,
I've been trying to install Carla and UE4 for a few days now, unfortunately without success.
I can't seem to progress past this error. It's not addressed in other issues either.
make PythonAPI was successful
I hope someone can help me.
Best regards

CARLA version: 0.9.15
Platform/OS: Windows

```
C:\carla>make launch
-[BuildUE4Plugins]: [Batch params]: --build --all
HEAD is now at 260273d Correct descriptor
-[BuildUE4Plugins]: STREETMAP has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins\"
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --generator "" --all
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Generator:          "Visual Studio 16 2019"
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
        1 Datei(en) kopiert.
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\carla\Build\" --zlib-install-dir "C:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "C:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\carla\Build\recast-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\carla\Build\" --toolset msvc-14.2 --version 1.80.0 -j 16
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\carla\Build\boost-1.80.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "C:\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "C:\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "C:\carla\Build\" --generator "Visual Studio 16 2019"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "C:\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
[... 169 lines cut ...]
  proj.lib(factory.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "sqlite3_libversion" in Funktion ""public: void __cdecl osgeo::proj::io::DatabaseContext::Private::open(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,struct projCtx_t *)" (?open@Private@DatabaseContext@io@proj@osgeo@@QEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@PEAUprojCtx_t@@@Z)".
  proj.lib(factory.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "sqlite3_libversion_number" in Funktion ""public: void __cdecl osgeo::proj::io::DatabaseContext::Private::open(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,struct projCtx_t *)" (?open@Private@DatabaseContext@io@proj@osgeo@@QEAAXAEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@PEAUprojCtx_t@@@Z)".
  proj.lib(factory.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "sqlite3_bind_double" in Funktion ""public: class std::list<class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > >,class std::allocator<class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > > __cdecl osgeo::proj::io::DatabaseContext::Private::run(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::list<struct osgeo::proj::io::SQLValues,class std::allocator<struct osgeo::proj::io::SQLValues> > const &,bool)" (?run@Private@DatabaseContext@io@proj@osgeo@@QEAA?AV?$list@V?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@V?$allocator@V?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@@2@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@7@AEBV?$list@USQLValues@io@proj@osgeo@@V?$allocator@USQLValues@io@proj@osgeo@@@std@@@7@_N@Z)".
  proj.lib(factory.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "sqlite3_column_count" in Funktion ""public: class std::list<class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > >,class std::allocator<class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > > __cdecl osgeo::proj::io::DatabaseContext::Private::run(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::list<struct osgeo::proj::io::SQLValues,class std::allocator<struct osgeo::proj::io::SQLValues> > const &,bool)" (?run@Private@DatabaseContext@io@proj@osgeo@@QEAA?AV?$list@V?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@V?$allocator@V?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@@2@@std@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@7@AEBV?$list@USQLValues@io@proj@osgeo@@V?$allocator@USQLValues@io@proj@osgeo@@@std@@@7@_N@Z)".
  proj.lib(factory.obj) : error LNK2019: Verweis auf nicht aufgelöstes externes Symbol "sqlite3_column_double" in Funktion ""public: class std::list<class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > >,class std::allocator<class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > > > __cdecl osgeo::proj::io::DatabaseContext::Private::run(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::list<struct osgeo::proj::io::SQLValues,class std::allocator<struct osgeo::proj::io::SQLValues> > const &,bool)" (?run@Priv [... cut ...]
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\lib\sqlite3.lib : warning LNK4272:Bibliothekcomputertyp "x86" steht in Konflikt mit dem Zielcomputertyp "x64"
  C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\lib\zlibstatic.lib : warning LNK4272:Bibliothekcomputertyp "x86" steht in Konflikt mit dem Zielcomputertyp "x64"
  C:\carla\Unreal\CarlaUE4\Plugins\CarlaTools\Binaries\Win64\UE4Editor-CarlaTools.dll : fatal error LNK1120: 45 nicht aufgelöste Externe
make: *** [CarlaUE4Editor] Fehler 6

```

siddharthjain38, 2024-03-22:
@evoevo07 I changed .\Util\BuildTools\BuildOSM2ODR.bat line 112 from

    cmake -G %GENERATOR% %PLATFORM%^
to

    cmake -G %GENERATOR% -A x64^
    
Than run make PaythonAPI and make launch again. before this clean make command or delete build folder.

evoevo07, 2024-03-24:
I believe I have identified my problem. The PythonAPI build failed during the make process because the dist folder is missing. Now, I have to resolve this issue.

evoevo07, 2024-03-27:
Not really a solution, but I switched to linux and now everything works fine.

Blyron, 2024-04-29:
It seems like sqlite is not being downloaded. Could you remove from Build folder sqlite-* folders and make launch again

Blyron, 2024-05-03:
Are you happy with the conclusion and happy to close this issue?

evoevo07, 2024-05-03:
@Blyron 
Thanks for supporting, but I can't test it, because I switched to Linux.
I think I will close the issue.

closed 2024-05-03 by evoevo07

thiagofcm, 2025-01-30:
Did you find any issue switching to linux? I am really think about switch to Linux, the amount of issues that I am facing on carla windows building is driving me crazy

siddharthjain38, 2025-01-30:
@thiagofcm till last month I was using carla on windows only, i was able to use it smoothly although I also faced multiple issue in the start. I also used latest maps and scenario-runner along with carla.