carla-simulator/carla #7283: Carla simulator 0.9.13 failed to launch

Anastasiawd, 2024-03-22:


Illumination720, 2024-04-07:
Same issue here and same GPU. Have tried many ways to make it run but failed. Appreciate for any suggestions. I ran it in the container with a carla version of 0.9.14 and 0.9.15 but both failed.

Anastasiawd, 2024-04-11:
> 
> 
> Same issue here and same GPU. Have tried many ways to make it run but failed. Appreciate for any suggestions. I ran it in the container with a carla version of 0.9.14 and 0.9.15 but both failed.

I have resolved this issue. It seems that the problem was caused by the server being unable to properly access the client's display to return images. A specific solution can be attempted by trying the solutions available online for X11 forwarding of server graphical interfaces.

MarcelPiNacy, 2024-05-03:
Hi,

I'm closing this issue as it seems completed, if this isn't the case feel free to reopen it.

Marcel

closed 2024-05-03 by MarcelPiNacy