carla-simulator/carla #7287: i installed Carla as per the instructions given on https://carla.readthedocs.io/en/0.9.15/build_windows/. but when i run  "C:\carsim\carla\Unreal\CarlaUE4\Binaries\Win64\CarlaUE4.exe"  its shows error.

hemant792, 2024-03-23:
i installed Carla as per the instructions given on https://carla.readthedocs.io/en/0.9.15/build_windows/. but when i run  "C:\carsim\carla\Unreal\CarlaUE4\Binaries\Win64\CarlaUE4.exe"  its shows error.
why is it happening? Everythings was done as per their webpage instructions. when i ran "make launch" command, it completed successfully but when it was loading unreal engine at the end, it got stuck at 95% for more than half an hour so i had to close it manually. and then when i run carlaUE4.exe i found this error.
[image]

_Originally posted by @hemant792 in https://github.com/carla-simulator/carla/discussions/7286_

lightlefter, 2024-04-10:
I still traped in "make launch "process [#7363](https://github.com/carla-simulator/carla/issues/7363).
when I saw your question , I  doubt if i should build carla 0.9.12 rather than 0.9.15.
too many problems in there.

xavisolesoft, 2024-04-12:
Hi,

The correct command to run CARLA is this one:
```bash
<UnrealEngine4_root>/Engine/Binaries/Linux/UE4Editor "<CarlaUe4_root>/CarlaUE4.uproject"
```

CarlaUE4.exe contains just he carla plugin and you will not be able to run it.