carla-simulator/carla #7296: Failed to start carla docker

EunhoCho, 2024-03-27:
CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04
Problem you have experienced: Tried to run carla docker on SSH with GUI
What you expected to happen: Run Carla properly.
Steps to reproduce:

I set up my GUI environment locally (Windows 11) and connected to the server where I want to run Carla.
Run Carla docker using the command below.
`docker run -it --name carla --privileged --rm --net=host -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw -v "$HOME/.Xauthority:/home/carla/.Xauthority:rw" carlasim/carla:0.9.15 ./CarlaUE4.sh -quality-level=Epic`

Then, it prints
```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
```

And it stops.

I tried many variances for docker commands, like adding `-e SDL_VIDEODRIVER=x11`, but nothing has worked.

EunhoCho, 2024-03-27:
Well, it works if I give up GUI.
`docker run -it --name carla --privileged --rm --gpus all --net=host -v /tmp/.X11-unix:/tmp/.X11-unix:rw carlasim/carla:0.9.15 ./CarlaUE4.sh -quality-level=Epic -RenderOffScreen`
works properly.

However, I wanted to see the GUI...

gawinwong, 2024-06-13:
You can use manual_control.py in examples folder to see pygame window of carla camera image

RParkerE, 2024-06-25:
@EunhoCho, it does not work giving up the GUI
[image]

Following the [steps to get NVIDIA support on Windows Docker](https://docs.docker.com/desktop/gpu/), and following your command as shown above in the snapshot, shows even when rendering off screen CARLA currently still throws the error

Abdallah-Hesham99, 2024-07-05:
I had similar problems and it kept crashing even though i installed cuda container, but it was solved using this command
 docker run --privileged --gpus all --net=host     -e DISPLAY=$DISPLAY     -e QT_X11_NO_MITSHM=1     -v /tmp/.X11-unix:/tmp/.X11-unix     --device /dev/snd     carlasim/carla:latest /bin/bash ./CarlaUE4.sh