carla-simulator/carla #7305: No rule to make target 'package' during build UE and Carla in Docker

xrkong, 2024-03-28:
CARLA version: 0.9.15
Platform/OS: Host OS Ubuntu 22.04, Docker OS Ubuntu 18.04
Problem you have experienced: 
```$ docker build -t carla -f Carla.Dockerfile . ```
[image]

I paste the logs for others to search.
```
BuildUtilsDocker.sh: Compiling FBX2OBJ...
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/carla/carla/Util/DockerUtils/fbx/build
[1/2] Building CXX object CMakeFiles/FBX2OBJ.dir/src/FBX2OBJ.cpp.o
../src/FBX2OBJ.cpp: In function 'int main(int, char**)':
../src/FBX2OBJ.cpp:232:49: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     gMatRoad     = CreateMaterial(lScene, "road");
                                                 ^
../src/FBX2OBJ.cpp:233:53: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     gMatSidewalk = CreateMaterial(lScene, "sidewalk");
                                                     ^
../src/FBX2OBJ.cpp:234:54: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     gMatCross    = CreateMaterial(lScene, "crosswalk");
                                                      ^
../src/FBX2OBJ.cpp:235:50: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     gMatGrass    = CreateMaterial(lScene, "grass");
                                                  ^
../src/FBX2OBJ.cpp:236:50: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
     gMatBlock    = CreateMaterial(lScene, "block");
                                                  ^
[2/2] Linking CXX executable FBX2OBJ
[0/1] Install the project...
-- Install configuration: ""
-- Installing: /home/carla/carla/Util/DockerUtils/fbx/../dist/FBX2OBJ
-- Set runtime path of "/home/carla/carla/Util/DockerUtils/fbx/../dist/FBX2OBJ" to ""
BuildUtilsDocker.sh: Success!
 ---> Removed intermediate container e890393ba3ed
 ---> 75d4750c148a
Step 6/9 : USER carla
 ---> Running in 7859fc2af9f2
 ---> Removed intermediate container 7859fc2af9f2
 ---> 7c84998f1f02
Step 7/9 : WORKDIR /home/carla
 ---> Running in 30836d7d2f4a
 ---> Removed intermediate container 30836d7d2f4a
 ---> 36438ef60d88
Step 8/9 : RUN cd /home/carla/ &&   make package &&   rm -r /home/carla/carla/Dist
 ---> Running in 636ba3b61013
make: *** No rule to make target 'package'.  Stop.
The command '/bin/sh -c cd /home/carla/ &&   make package &&   rm -r /home/carla/carla/Dist' returned a non-zero code: 2
```

What you expected to happen:
Keep building the image.

Steps to reproduce: 
```$ docker build -t carla -f Carla.Dockerfile . ```
Modified Carla.Dockerfile shown as follow,
```
FROM carla-prerequisites:latest

ARG GIT_BRANCH

USER root
WORKDIR /home/carla

RUN cd /home/carla/ && \
  if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git; \
  else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi && \
  cd /home/carla/carla && \
  ./Update.sh && \
  make CarlaUE4Editor && \
  make PythonAPI && \
  make build.utils 

USER carla
WORKDIR /home/carla

RUN cd /home/carla/ && \
  make package && \
  rm -r /home/carla/carla/Dist

WORKDIR /home/carla/carla
```

Other information (documentation you consulted, workarounds you tried):
[Build Unreal Engine and CARLA in Docker](https://carla.readthedocs.io/en/latest/build_docker_unreal/)

PatrickPromitzer, 2024-04-11:
The command "make package" is calling other make commands to build the image.
For that reason, I would look up the sub commands in the make file

https://github.com/carla-simulator/carla/blob/dev/Util/BuildTools/Linux.mk
`package: CarlaUE4Editor PythonAPI`
You can try if the sub commands work and go from there.

The real error message in between the "[655/659]" lines, because Carla is building with multiple Threads.
If one Thread has an error, the other Threads finish the run, and the error message will get lost in between. 

I only made a image for 0.9.13 and 0.9.14, and still have to bug-fix the Dockerfile for 0.9.15
I may can help more if I have an 0.9.15 image.

MarcelPiNacy, 2024-04-26:
Hi,

could you provide the full log along with the file "/root/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt"?

Marcel

xrkong, 2024-04-27:
> Hi,
> 
> could you provide the full log along with the file "/root/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt"?
> 
> Marcel

Hi Marcel,

I didn't find that folder in that path. But I find another Log.txt in here,
```
/home/carla/UE4.26/Engine/Programs/UnrealBuildTool/Log.txt
```

[Log.txt](https://github.com/carla-simulator/carla/files/15136786/Log.txt)

Thanks for your comment!

Ray

MarcelPiNacy, 2024-04-27:
Hi again,

The log you provided is from the build step, but we are actually interested in the cook part. Please check whether there is a log in the corresponding folder for UE's AutomationTool.

Thank you,
Marcel

xrkong, 2024-05-03:
Hi Marcel,

Thank you for your attention. 
I went into the container, ran `make package` manually and found the file you mentioned. 

[UBT-UnrealPak-Linux-Development.txt](https://github.com/carla-simulator/carla/files/15200135/UBT-UnrealPak-Linux-Development.txt)
[UBT-CarlaUE4-Linux-Shipping.txt](https://github.com/carla-simulator/carla/files/15200137/UBT-CarlaUE4-Linux-Shipping.txt)
[Log.txt](https://github.com/carla-simulator/carla/files/15200138/Log.txt)

Regards,
Ray

xrkong, 2024-05-03:
Hi, 
The build part seems need `root`, but the cook part says `Refusing to run with the root privileges.` which is confused me.
Regards,
Ray