carla-simulator/carla #7308: How to get longitudinal and lateral velocities?

Anhar2, 2024-03-31:
I'm using CARLA 0.9.14 "with leaderboard v2" on Ubuntu 20 Focal Fossa

In a scenario, I use get_velocity() to return the x and y velocities, how to transform these values into longitudinal and lateral velocities as I need them to calculate RSS measurements such as MSDV?

glopezdiest, 2024-05-10:
@Anhar2 You can't really do that unfortunately. The only solution is to create a new function that instead of returning the speed, it returns the velocity vector. That means:

1) Adding a 4th map similar to the `CarlaDataProvider._actor_velocity_map`, `CarlaDataProvider._actor_location_map` and `CarlaDataProvider._actor_transform_map` ones
2) Update it on the `on_carla_tick()`
3) Create the getter

closed 2024-05-10 by glopezdiest

Anhar2, 2024-05-13:
> @Anhar2 You can't really do that unfortunately. The only solution is to create a new function that instead of returning the speed, it returns the velocity vector. That means:
>
> 1. Adding a 4th map [...]

Thank you for Your response!