carla-simulator/carla #7311: Custom Vehicle Head Lights and Adjustable vehicle light parameters through Python API as future features

AyumuOshiro-RCI, 2024-04-02:
Hi, Thank you for developing a great simulator.

I am interested in utilizing CARLA specifically for car headlights simulation with MATLAB/Simulink models. 
In order to imitate ADB(Adaptive Driving Beam) and AFS(Adaptive Front Lighting System) in CARLA, I'm very happy that if CARLA has a feature such as adjusting the head lights parameters like the Light intensity through Python API as same as the street light does.

Do you plan to add such a feature for a future release?

[image]

 
Also, maybe it is already possible but I would like to add head lights onto my custom vehicle added in carla. If it is possible, I would appreciate if anyone helps me to do this.

GoodarzMehr, 2024-04-04:
The Python API only lets you change the light state (turn on/off low beam, high beam, fog lights, etc). AFAIK you have to write your own C++ code to change the intensities.

You can add lights to a custom vehicle (I've done it successfully for a Mustang I imported), the full instructions are in the tutorial for adding a new vehicle but basically, you first need to create a mesh shaped like the lights, then apply the proper UV texture, export it to .fbx, import it in UE4, apply the light material to the mesh, add the mesh to the blueprint and tag it as emissive, and finally add spot lights similar to the ones in existing blueprints.

gsmario, 2024-04-12:
Hello! You can follow the documentation to add new lights to your vehicles: 
https://carla.readthedocs.io/en/latest/tuto_content_authoring_vehicles/#lights
Regarding the lights intensity, for now, you can only change the streetlights, if you want to change the vehicle intensity, you should modify the vehicle blueprint, rightnow you can't use the api for that, sadly.

gsmario, 2024-05-24:
I'm closing the issue, since the problem seems solved.
Thanks for your time.

closed 2024-05-24 by gsmario