carla-simulator/carla #7324: carla runs without the GPU

Morizb, 2024-04-04:
Hi, I installed carla in a conda environment on a remote server, installed openGL in a docker container, and used ssh to remotely connect to Mabaxterm and configure X11-forwarding for visualization.

Execute the following command to run carla: CUDA_VISIBLE_DEVICES=0 . /carla/CarlaUE4.sh -prefernvidia -opengl -carla-server -benchmark -fps=5 -nosound --world-port=15302
[image]

[image]

But carla runs without GPU, resulting in rendering no environment.
[image]

Can you please tell me how to solve this problem？

Blyron, 2024-04-10:
We recommend to use vulkan, have you tried with it?

Morizb, 2024-04-16:
> We recommend to use vulkan, have you tried with it?

Thanks for your reply.
I downloaded vulkan, but it reports an error: Cannot find a compatible yulkan driver (ICD)
[image]

The driver version on the server is: NVIDIA-SMI 515.65.01 Driver Version: 515.65.01 CUDA Version: 11.7
This seems to be because there is a fixed version requirement for the vulkan 1.3 driver, but I don't have permission to change the nvidia driver version on the school server.
Is there any other way to resolve this issue please?

joel-mb, 2024-04-19:
Hi @Morizb 
It seems that the loader is not finding your Vulkan drivers. Could you try sharing with your docker container the `icd` JSON files:

```bash
--volume=/usr/share/vulkan/icd.d:/usr/share/vulkan/icd.d:ro
```

joel-mb, 2024-04-29:
Closing the issue. Please, feel free to reopen the issue if needed.

closed 2024-04-29 by joel-mb

IranQin, 2024-08-20:
> Hi, I installed carla in a conda environment on a remote server, installed openGL in a docker container, and used ssh to remotely connect to Mabaxterm and configure X11-forwarding for [...]

15/5000

Hello, may I ask how you finally solved this problem？