carla-simulator/carla #7328: Carla fails to install in Python 3.11 venv

mimi89999, 2024-04-04:
CARLA version: 0.9.15
Platform/OS: Debian trixie
Problem you have experienced: The python lib fails to install in a Python 3.11 venv. I also noticed that there is no wheel for Python3.11 on pypi
What you expected to happen: The module should install
Steps to reproduce:
In a Python 3.11 venv:
```
$ pip install carla
Collecting carla
  Downloading carla-0.9.5.tar.gz (3.9 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [21 lines of output]
      Traceback (most recent call last):
        File "/tmp/CARLA_0.9.15/PythonAPI/examples/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/tmp/CARLA_0.9.15/PythonAPI/examples/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/CARLA_0.9.15/PythonAPI/examples/.venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-8cpj1uij/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 325, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-8cpj1uij/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 295, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-8cpj1uij/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 487, in run_setup
          super().run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-8cpj1uij/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 114, in <module>
        File "<string>", line 31, in get_libcarla_extensions
      AttributeError: module 'platform' has no attribute 'dist'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```
Other information (documentation you consulted, workarounds you tried):

mimi89999, 2024-04-05:
The problem is that there is no source dist archive for `0.9.15`, so it downloads the last source archive that it can find and that is `0.9.5`.

davidscientistcom, 2024-04-08:
Hello,

According to the library’s download page, the CARLA client is already compatible with Python 3.10, which is the version I am currently using.

Best regards,

joel-mb, 2024-05-10:
@mimi89999 Right now CARLA is supporting up to python3.10 (https://pypi.org/project/carla/#files). We will work for bringing newer python version but for the moment you should downgrade your python version to 3.10.

closed 2024-05-10 by joel-mb