carla-simulator/carla #7329: Can two instances of Carla run on the same machine?

abol-karimi, 2024-04-04:
I'm trying to run multiple Carla jobs on an HPC cluster. Sometimes two or more jobs get assigned on the same node but on different GPUs, and when that happens both Carla instances crash. I've made sure that the ports (rpc, steaming, traffic manager) are different for different instances, but that didn't help.

GoodarzMehr, 2024-04-05:
I've had the same experience. The problem seems to come from both instances having access to the same shared memory and messing around with some shared object, which causes both to segfault.

abol-karimi, 2024-04-05:
I'm running the instances inside Singularity containers, but that doesn't help either.

GoodarzMehr, 2024-04-05:
Same here, the containers share the underlying resources so it has no effect.

abol-karimi, 2024-04-05:
Do you know which parts of Carla use such shared memories? I'd like to take a closer look.

GoodarzMehr, 2024-04-05:
For me most segfaults have occured when trying to destroy an actor (i.e. vehicle, sensor, walker, or walker controller) so I'd probably take a look at how CARLA handles those.

xavisolesoft, 2024-04-12:
I share the CARLA multi gpu documentation, let me know if this solves yout issue:
https://carla.readthedocs.io/en/latest/adv_multigpu/#carla-multi-gpu-feature

closed 2024-04-12 by xavisolesoft

abol-karimi, 2024-05-08:
I was able to fix the port isolation as well. I just use the Apptainer's `--net --network=none` option to isolate the network of each container. The only caveat with this solution is that the container will not have access to the internet. For example, timm won't be able to download from the Hugging Face hub, but I can run it once without the network isolation to let timm cache on my home directory.

abol-karimi, 2024-05-08:
> For me most segfaults have occured when trying to destroy an actor (i.e. vehicle, sensor, walker, or walker controller) so I'd probably take a look at how CARLA handles those.

I run my CARLA scenarios using SCENIC and haven't had that problem. I built CARLA with the Debug config to get more detailed logs. The source of my crashes happened to be that some nodes on the cluster have newer versions of NVIDIA drivers and Apptainer's `--nv` option was not working properly. The system admin updated Apptainer to 1.3.0-1 and problem solved!