carla-simulator/carla #7330: Building new actor.h and actorcontroller.h for PythonAPI

LeonardoVezzani, 2024-04-05:
I'm trying to implement new kind of actors and their relative controller to be accessed and controlled from the client using the PythonAPI.

I'm building on Win10/Win11 CARLA 0.9.15.

This is what I did so far:

Copying Walker, WalkerBase, WalkerControl, WalkerController I created the classes Dashboard, DashboardBase, DashboardControl, DashboardController.

I proceeded doing the following:

In /LibCarla/source/carla:
     in /client I created the Dashboard.cpp/.h
     in /client/detail/client.cpp I created the `_pimpl->Async` for the DashboardControl
     in /client/detail/simuklator.h I created the `applyControlToDashboard` function
     
     in /rpc/command.h I created the `struct ApplyDashboardControl` and I added it to the `CommandType = boost::variant2::variant <>`
     in /rpc I created the DashboardControl.h that is basically the same of WalkerControl.h with changed names
     in /sensor/data/ActorDynamicState.h I implemented the `PackedDashboardcontrol` and all the relative functions and metods
     
In PythonAPI/carla/source/carla/
    actor.cpp I created the function `ApplyDashboardControl` and the relative `.def`
    commands.cpp i created the `class_<cs::Command::ApplyDashboardControl>` and all the relative `.def` and `implicit_convertible` as done for the WalkerControl
    control.cpp I created the `std::ostrem &operator`
    
in Ureal/CarlaUE/Plugins/Carla/Source/carla
    /actor/ActorData.cpp and CarlaActor.cpp I copied all the WalkerController adapting it for the code for the DashboardController (just copied it and changed Walker to Dashboard)
    Added the folder /dashboard where Dashboard, DashboardControl, Dashboardcontroller are ( which are copies of the relative walker classes with different names)
    in /game/FrameData.cpp I added the relative `FDashboardControl`
    in /sensor/WorldObserver.cpp I added the sleif cases to include the dashboardcontrol 
    in Server/CarlaServer I made the `BIND_SYNC` for all the DashboardController to work. 
    
The doc has been updated automatically during the build as the new classes appear in it
     
After all this work the I made several builds with successful results but in the PythonAPI and in the carla library that I import in my client scripts I can only access Dashboards but I cannot access the DashboardControl methods.

At this point I don't understand what to do to make it work and even the CARLA devs couldn't help in solving this issue.
I guess I might share the files but only if it is strictly necessary as I'd like to publish the results in a journal that would require anonymity.

MarcelPiNacy, 2024-05-03:
Hi,

Sorry for the delay, do you still have this issue?
If so, ping me on the CARLA discord server.

Marcel

LeonardoVezzani, 2024-05-06:
Hi Marcel,
Thank you for your interest and excuse me for my delay but I fell seek last week.

Yes the issue persist:
I still can’t create a new actor controller for my new actor. I was thinking to try it all over again with a more simplistic approach but my lab recently got new computer, so it took some time to move all the active projects and couldn’t get back to it sooner than today.

I’d really appreciate if you could share some tip or insights on the topic.
Thank you

L.V.