carla-simulator/carla #7331: CARLA is dead, long live CARLA

davebiagioni, 2024-04-06:
I love CARLA, used it extensively for research a few years back. But it's time for the maintainers to declare that it has passed on to the autonomous afterlife.

Why do I say this? Perusing the backlog, it's obvious this project has no active support and no plans to change that fact.

See: #6903  #7213  -> No support of python > 3.8 (which is already dead itself: https://devguide.python.org/versions/#versions) 

Also, compare the oldest issues (of which there are 1.1k open) with the newest. The oldest had assignees and active commentary. The last several months are crickets.

Again, love this project, love what it was trying to do. But do the right thing and announce the death of CARLA so everyone can move on in whatever way they deem appropriate.

davebiagioni, 2024-04-06:
Had a follow on thought, we should turn this issues tab into a tribute to CARLA! Here is an overhead video I made a few years ago, driving the vehicle with model predictive control (the "2d bicycle model").

https://github.com/carla-simulator/carla/assets/7086434/b3e99f52-5521-4afe-8616-0893dbbacbdb

Demo your CARLA magic here!

Illumination720, 2024-04-07:
RIP LOL... Cheers!

davidscientistcom, 2024-04-08:
Hello,

I understand your concerns regarding the support for CARLA. However, I would like to point out that, according to the library’s download page, the CARLA client is already compatible with Python 3.10, which is the version I am currently using. This indicates that the project continues to receive updates and support for newer versions of Python.

Best regards,

https://pypi.org/project/carla/#files

PatrickPromitzer, 2024-04-11:
Hi,
I know what you mean.
Installing the Carla source version is not easy, and if one part is wrong, it doesn't work.
The packages it uses are old.
For example:
Carla is using clang-7 (release 2018) and the current version is clang-18?
Ubuntu 22.04 stopped the support of clang-7 a year ago.

The Carla fork from the Unreal Engine Editor makes it more complex as it should be, too.
It would be better to have some plugins (Carla sensors, the python API, ...), and some maps in a project.

I would not trust the issue number.
I started to answer some of the issues, but a only few were closed.

The idea itself (and if you only work with the builds) is not bad, but the updates and fixes Carla is getting are very slow (if they get fixed).

Version 0.9.14 got a nice update to see material data of the objects with the RGB (camera) sensor, but it only works limited (it breaks if you use too many). It has the name listen_to_gbuffer
Version 0.9.15 broke the listen_to_gbuffer function and freezes Carla if used.

I could go on for some time, but I don't think it is worth writing everything down if the important people don't read it.

I still have to use Carla for some time, and I hope it will change in a better way.

GoodarzMehr, 2024-04-14:
I think it's natural for the project to have more bugs and issues as it gets bigger and bigger and adds new features with each new version. I don't think the CARLA dev team is large and so it becomes a matter of prioritizing either adding new things or addressing all the bugs, and they have clearly chosen the latter, since they have e.g. added three very large maps and focused on increasing the variety of assets (e.g. by integrating it with Nvidia's Omniverse) as well as upgrading to Unreal Engine 5 in recent versions, which may also be what those who financially support CARLA are looking for. I have had a lot of issues with CARLA and working with it has been frustrating many times (see e.g. #7163 and #7191), but the project is open source and people can resolve a lot of issues on their own.

Side note, and I hope you don't take this the wrong way, say the dev team announces that CARLA has moved on to greener pastures, now what?

davebiagioni, 2024-05-04:
I'm just saying, RIP. It was good while it lasted.

closed 2024-05-04 by davebiagioni

platdesk, 2024-05-10:
> Hi, I know what you mean. Installing the Carla source version is not easy, and if one part is wrong, it doesn't work. The packages it uses are old. For example: Carla is using clang-7 (release 2018) [...]
I came up with the same problem in 0.9.15 and remain unsolved. May i know is there any solutions now.Thangks.