carla-simulator/carla #7332: The effect of Chrono doesn't work

donghaiwang, 2024-04-08:
Using`make launch ARGS="--chrono"` to start the server, the effect of running  [`manual_control_chrono.py`](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/examples/manual_control_chrono.py) looks the same as that of running [`manual_control.py`](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/examples/manual_control_chrono.py) and press `Ctrl + o` to enable Chrono. Is there any difference in effect?

CARLA version: 0.9.15
Platform/OS: Win10

saippua, 2024-05-06:
I am also having trouble getting Chrono to work on windows.

I can see that chrono is built and installed in `carla/Build/chrono-install` and executing the binaries seems to work, but chrono will not enable in the `manual_control_chrono.py` when pressing `Ctrl+o`. I've checked that the relevant line `world.player.enable_chrono_physics(...)` is being executed, but it just doesn't do anything.

I've also tested this on Ubuntu 20.04 Linux with the exact same setup and it works out of the box. Chrono is correctly enabled when pressing `Ctrl+o`

donghaiwang, 2025-01-23:
When turning at high speed, turning the steering wheel all the way will cause the car to flip over. For animation effects, please see the link:  https://openhutb.github.io/carla_doc/tuto_G_chrono/

closed 2025-01-23 by donghaiwang