carla-simulator/carla #7338: Exporting a map on Windows

NnaYelsel, 2024-04-08:
CARLA version: 0.9.15
Platform/OS: Windows11

Hello,

After making a new map from .xodr + .fbx, I did as mentioned in https://carla.readthedocs.io/en/latest/tuto_content_authoring_maps/#importing-your-road-network-into-carla with the command:

        make package ARGS="--packages=MyMAp"

Then on /Build/UE4Carla/ I got an archive which is looking like AdditionalMaps_0.9.15 but the folder CarlaUE4/Content/Carla/Maps is missing and the umap is not in CARLA_ROOT/Unreal/CarlaUE4/Content/Carla/Maps

Did I miss something ?

I tried also "Exporting a map as part of a complete CARLA package" but that didn't work either.

Best Regards

NnaYelsel, 2024-04-11:
Hi, this is my logs:

-[Package]: [Batch params]: --ue-version 4.26 --packages=ElevationHouse
Parsing packages...
Preparing environment for cooking 'ElevationHouse'.
Cooking package 'ElevationHouse'...
  - prepare
Prepare cooking of package: ElevationHouse
[2024.04.11-09.26.35:345][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'AllDesktop'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_ASTC'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_DXT'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_ETC2'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'AndroidClient'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_ASTCClient'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_DXTClient'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_ETC2Client'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_Multi'
[2024.04.11-09.26.35:356][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Android_MultiClient'
[2024.04.11-09.26.35:367][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'IOSClient'
[2024.04.11-09.26.35:367][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'IOS'
[2024.04.11-09.26.35:379][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Lumin'
[2024.04.11-09.26.35:379][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'LuminClient'
[2024.04.11-09.26.35:382][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'MacNoEditor'
[2024.04.11-09.26.35:387][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Mac'
[2024.04.11-09.26.35:391][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'MacClient'
[2024.04.11-09.26.35:395][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'MacServer'
[2024.04.11-09.26.35:401][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'TVOSClient'
[2024.04.11-09.26.35:402][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'TVOS'
[2024.04.11-09.26.35:405][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'WindowsNoEditor'
[2024.04.11-09.26.35:410][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'Windows'
[2024.04.11-09.26.35:414][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'WindowsClient'
[2024.04.11-09.26.35:418][  0]LogTargetPlatformManager: Display: Loaded TargetPlatform 'WindowsServer'
[2024.04.11-09.26.35:418][  0]LogTargetPlatformManager: Display: Building Assets For Windows
[2024.04.11-09.26.35:422][  0]LogAudioDebug: Display: Lib vorbis DLL was dynamically loaded.
[2024.04.11-09.26.35:454][  0]LogShaderCompilers: Display: Using Local Shader Compiler.
[2024.04.11-09.26.35:895][  0]LogDerivedDataCache: Display: Max Cache Size: 512 MB
[2024.04.11-09.26.35:912][  0]LogDerivedDataCache: Display: Loaded Boot cache: ../../../../Public_CARLA/CARLA_source_code/carla/Unreal/CarlaUE4/DerivedDataCache/Boot.ddc
[2024.04.11-09.26.35:916][  0]LogDerivedDataCache: Display: Performance to ../../../Engine/DerivedDataCache: Latency=0.01ms. RandomReadSpeed=2021.50MBs, RandomWriteSpeed=245.47MBs. Assigned SpeedClass 'Local'
[2024.04.11-09.26.36:066][  0]LogModuleManager: Warning: ModuleManager: Unable to load module 'ExampleDeviceProfileSelector'  - 0 instances of that module name found.
[2024.04.11-09.26.36:755][  0]LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
[2024.04.11-09.26.36:755][  0]LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
[2024.04.11-09.26.36:758][  0]LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
[2024.04.11-09.26.36:766][  0]LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
[2024.04.11-09.26.37:124][  0]LogAudioCaptureCore: Display: No Audio Capture implementations found. Audio input will be silent.
[2024.04.11-09.26.37:124][  0]LogAudioCaptureCore: Display: No Audio Capture implementations found. Audio input will be silent.
[2024.04.11-09.26.38:723][  0]LogTemp: Warning: Found 0 tiles
[2024.04.11-09.26.38:870][  0]LogInit: Display: 
[2024.04.11-09.26.38:870][  0]LogInit: Display: Warning/Error Summary (Unique only)
[2024.04.11-09.26.38:871][  0]LogInit: Display: -----------------------------------
[2024.04.11-09.26.38:871][  0]LogInit: Display: RenderDocPlugin: Warning: Unable to initialize the plugin because no RenderDoc libray has been located.
[2024.04.11-09.26.38:871][  0]LogInit: Display: LogConsoleManager: Warning: Setting the console variable 'r.Streaming.PoolSize' with 'SetByScalability' was ignored as it is lower priority than the previous 'SetByProjectSetting'. Value remains '4000'
[2024.04.11-09.26.38:871][  0]LogInit: Display: LogModuleManager: Warning: ModuleManager: Unable to load module 'ExampleDeviceProfileSelector'  - 0 instances of that module name found.
[... cut ...]
[2024.04.11-09.26.53:474][  0]LogShaderCompilers: Display: Shaders left to compile 0
[2024.04.11-09.26.53:588][  0]LogHttp: Display: cleaning up 0 outstanding Http requests.
[2024.04.11-09.26.54:050][  0]LogContentStreaming: Display: There are 1 unreleased StreamingManagers
Copying files to 'ElevationHouse'...
        1 file(s) copied.
Packaging 'ElevationHouse'...

7-Zip 19.00 (x64) : Copyright (c) 1999-2018 Igor Pavlov : 2019-02-21

Open archive: C:\Users\Public\Public_CARLA\CARLA_source_code\carla\Build\UE4Carla\ElevationHouse_0.9.15-dirty.zip

NnaYelsel, 2024-04-11:
I think that the make import is made to create map etc... in map_package folder but ake package ARGS="--packages=MyMAp" is made to be called with your map on Content/Carla/Maps isn't it ?

Finally could make package after manually moving the content of map_package into Carla/Maps, but my package still have the folder map_package with the terrain on it. It' not really clean ... and my spawn point are not compiled...

Blyron, 2024-04-12:
Hello! 
When making package Unreal make paths from /Content/ to /Game/ thats why you need to place you map inside /Content/Carla/Maps. Your spawn points will be placed in the map if you placed them in editor mode.

NnaYelsel, 2024-04-12:
Hi @Blyron, 
Thank you for the answer 🙏

The fact is that after putting my .xodr and .fbx in "Import" folder and calling "python ...../import.py", my map is placed in "map_package" folder, that contains these 3 folders:
- Config
- Maps/MyMap
- Static (with Road/MyMap, RoadLine/MyMap and Terrain/MyMap)

If I manually cut/paster the content of Maps/MyMap into /Content/Carla/Maps and the content of Static in /Content/Carla/Static, when opening my map on UE editor, the map will be almost empty because some "PropNode" are still looking for the content of map_package/Static/Stuffs 

Example:

`Failed to load /Game/map_package/Static/Terrain/ElevationSimple/ElevationSimple_f_shaped_barrier_mesh_PropNode.ElevationSimple_f_shaped_barrier_mesh_PropNode Referenced by ElevationSimple_f_shaped_barrier_mesh_PropNode`

If I only copy/paste it, then it's compiling but a map_package is created in my package to basically contains all the static stuff, i.e Terrain and roads. I.e MyMap_0.9.15-dirty\CarlaUE4\Content contains Carla and map_package and if I'm deleting map_package, I don't have my terrain anymore. 
It's not clean but that's working 😀. 

I wonder if it could be a good idea to modify the python code import.py with 
```
def main():
    argparser = argparse.ArgumentParser(description=__doc__)
    argparser.add_argument(
        '--package',
        metavar='P',
        **default='map_package', ---> change in favor of default='Carla'**
        help='Name of the imported package')
    argparser.add_argument(
        '--no-carla-materials',
        action='store_false',
        help='user Carla materials')
    argparser.add_argument(
        '--json-only',
        action='store_true',
        help='Create JSON files only')
```
in order to have directly everything in the Carla folder ? 

**Most important question:**
By the way, when opening my map just imported from the .fbx/.xodr couple, I have some spawn point already defined. Do you mean that I should delete them all and replace them manually from Carla Editor ? 
What is the best way to import them ? Going to Carla C++ Classes > Carla > Vehicle > VehicleSpawnPoint and place them ? that's all ?

Blyron, 2024-04-12:
It should be placed automatically afaik. If not, Yes, just place a VehicleSpawnPoint in points are suitable for you. 

About import stuff I guess is because how references works in UE. If you are using Unreal Editor I suggest you to create a Level like our maps then place all your stuff, and then add the map in MapsToCook. Then you can modify default maps so the one which is loaded is yours. I think for now our Import is ok, but we will look to improve it in the future

closed 2024-04-12 by NnaYelsel