carla-simulator/carla #7341: [Python agents] `set_destination` has unused parameter and `_tailgating` bugged

Daraan, 2024-04-09:
Carla 0.9.15 and before:

### The problem:

The `set_destination` function currently is illogical as it only uses the `start_location` for boolean evaluation and not as a location anymore.

https://github.com/carla-simulator/carla/blob/cead6d1b05e89ad84df80984f62daba57bb50e55/PythonAPI/carla/agents/navigation/basic_agent.py#L141
https://github.com/carla-simulator/carla/blob/cead6d1b05e89ad84df80984f62daba57bb50e55/PythonAPI/carla/agents/navigation/basic_agent.py#L151-L156

### Manifestation:

The only native example where this is used is in the BehaviorAgent tailgating evasion maneuver:

https://github.com/carla-simulator/carla/blob/cead6d1b05e89ad84df80984f62daba57bb50e55/PythonAPI/carla/agents/navigation/behavior_agent.py#L119-L120

In my experiments this is broken as the agent makes a U-Turn to reach a weaypoint that is far behind the current location.

Daraan, 2024-06-21:
I think this is partially fixed with #7342, the BehaviorAgent will now not longer append the current location to the queue, to which it tries to come back later.

However as the queue is cleaned there is no plan beyond this. This might be similar to the original behavior when `set_destination` and `_tailgating` where still in sync but I have not researched this.

M-Colley, 2026-10-26:
Draft PR #10009 proposes the remaining `ue58-dev` tailgating-route fix, preserving the route destination while checking usable adjacent lanes. It includes offline agent regressions. The earlier `start_location` behavior is a separate change already handled by #7342; live Traffic Manager/agent validation remains unrun.

closed 2026-11-12 by glopezdiest