carla-simulator/carla #7343: Performance issue across Different PCs

ysrhee6199, 2024-04-09:
CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04

When running CARLA's manual_control.py on three different PCs, there's a significant performance difference observed.

CPU: 5600x
Motherboard: Gigabyte B450M
RAM: 32GB
Graphics Card: RTX 4060ti

CPU: i9-13900
Motherboard: MSI MAG B760M
RAM: 64GB
Graphics Card: RTX 3080ti

CPU: i9-10900xe
Motherboard: WS X299 SAGE/10G
RAM: 128GB
Graphics Card: RTX 3090

While the first and second PCs show very good performance with server FPS ranging from 100 to 120 FPS when executing manual_control.py, the third PC comparatively shows much poorer performance, with FPS ranging from 40 to 60.

I'm curious about the reasons for this performance discrepancy.

xavisolesoft, 2024-04-12:
Hi,

We suspect that the difference of FPS is due hardware specifications of each machine.

Anyway, would be useful for us if you could share profiling to dive deep more in your case. I share you the documentation about how to profile:
https://docs.unrealengine.com/4.27/en-US/TestingAndOptimization/PerformanceAndProfiling/

Best Regards

MarcelPiNacy, 2024-05-03:
Hi,

Also, if you're running this on the UE editor directly make sure to uncheck the "Use less CPU while in background" option.

Marcel