carla-simulator/carla #7381: Creation of new bones in the General4WheeledVehicleSkeleton

eindino1, 2024-04-10:
Hi,
I'm actually try to create new bones for **[4-wheeled Vehicles]**, this new bones is for make a doors opening or other "animation" on some part of the vehicule. 

My question is : Their is a clean way to do it and integrate well in the current version of carla. 
I'm following this tuto https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/

Blyron, 2024-04-10:
Hello!

For this purpouse you will need to create a new skeleton to Unreal Engine, create new BP, create the animation BP, bind the input from PythonAPI to LibCarla to CarlaUE to AnimationBlueprint. Then add new BP to the VehicleFactory. Is not trivial task. What you can do is open UE Editor  to import the vehicle without doors and then import doors as separate meshes and try to duplicate one of our cars and swap meshes in UE Editor

Madecu, 2024-04-12:
Hi @eindino1 
There's another way to achieve what you're asking for, which is the method we've used in CARLA to open the doors of some vehicles: Having the doors as separate assets and assembling them within the BP, then moving them via code through rotations and constraints. You can refer to vehicle Blueprints such as the Nissan Patrol or Charger for guidance

closed 2024-05-03 by Madecu