carla-simulator/carla #7387: carla can generate anomalous data

zeroyang0809, 2024-04-10:
i want to Detect the matching and authenticity of sensor data
 • Input speed, acceleration, throttle, brake, GPS changes
 • Detect anomalies such as
 ◦ GPS change when speed is 0
 ◦ Speed change when acceleration is 0
 ◦ Throttle and brake increase at the same time

can carla generate some data like this ?

PatrickPromitzer, 2024-04-10:
You can set and get parameters, but it will not be 100% the same as real live ground-proof data.

The parameters  that are similar to the one you need are
- get_acceleration
- get_angular_velocity
- get_location
- get_transform
- get_velocity

https://carla.readthedocs.io/en/latest/python_api/#carlaactor

zeroyang0809, 2024-04-12:
thanks for your reply！
I am curious that can Carla generate some anomalous data based on specific guide or random way ? 
I do not find this approach.

PatrickPromitzer, 2024-04-12:
It would help if you explain what you want to do.

Carla is only a simulation, and if you want to generate data, you need to tell Carla what to do.
You can use ScenarioRunner to add vehicles that drive along the road with an openSCENARIO file.
Carla has a physic engine (Carla is an Unreal Engine project), and if you let the car drive, Carla will calculate how the vehicle moves, based on different variables.

If you want to have some random behavior, you could play around with the acceleration, steering angle, ...  to get different kinds of data.

closed 2024-05-03 by MarcelPiNacy

PatrickPromitzer, 2024-07-02:
1.) Carla gives the ground proof data of the Transform (location + rotation), and to have changes which are not the right Transform data, you would have to add a filter to add noise. 
```
location_x = location.x + random_noice
```
2.) For throttle, brake, ..., you can use the VehicleControl
https://carla.readthedocs.io/en/0.9.15/python_api/#carla.VehicleControl
3.) you can use the [generate_traffic.py](https://github.com/carla-simulator/carla/blob/dev/PythonAPI/examples/generate_traffic.py) and look how the Vehicles behave. 

In the end, you have to look yourself if the data fits to your purpose.
You can look in the [examples]( https://github.com/carla-simulator/carla/tree/dev/PythonAPI/examples) folder to get a base script for controlling Carla.