carla-simulator/carla #7393: Carla API version  and simulator API version problem

zeroyang0809, 2024-04-11:
In the past, i use carla0.9.11. 
I want to use carla 0.9.15 version recently.
However, when i run the config.py to change the town map.
It looks like that it meet some version problem

how to handle the problem like this?
ps: i install the carla by download the https://github.com/carla-simulator/carla/releases/tag/0.9.15/and https://github.com/carla-simulator/carla/releases/tag/0.9.11/
thanks

[image]

FriedrichZimmer, 2024-04-11:
I think you should also update the carla library in python by pip after updating the server to 9.15

Aperiss, 2024-04-19:
you need to update the CARLA client installation, first:

# Inside the deliverable folder run:
pip3 uninstall carla
pip3 install -r requirements.txt

Then you install the updated requirements and call make PythonAPI

closed 2024-07-29 by Blyron