carla-simulator/carla #7394: Collision sensor is sometimes not working

glopezdiest, 2024-04-11:
Following on [this issue](https://github.com/carla-simulator/carla/issues/7025), the CollisionSensor has an issue where it can sometimes fail with the detection of collisions

glopezdiest, 2024-04-11:
After some quick tests, this issue seems to always happen with invincible pedestrians as shown with the script below. Additionally, the direction of the actor that is collided with seems to also affect its detection, but I haven't managed to create a script that fully reproduces it

[coll_test.zip](https://github.com/carla-simulator/carla/files/14944300/coll_test.zip)

AOOOOOA, 2024-04-16:
> After some quick tests, this issue seems to always happen with invincible pedestrians as shown with the script below. Additionally, the direction of the actor that is collided with seems to also [...]

Hi, 

The issue also happens with vehicles.
Sorry that I forgot a detail. I replace the "tesla.model3" in FollowLeadingVehicle.xsoc file as "**vehicle.bh.crossbike**" which also belongs to _vehicle_ type in Carla(https://carla.readthedocs.io/en/latest/bp_library/).
Attached is the modified FollowLeadingVehicle.xsoc for you to refer, please have a try.
[FollowLeadingVehicle.xosc.zip](https://github.com/carla-simulator/carla/files/14986138/FollowLeadingVehicle.xosc.zip)

Following the code and the command provided in the FollowLeadingVehicle part, you can get the unreported collision events as the video shows in the corresponding issue:
https://github.com/carla-simulator/carla/issues/7025#issuecomment-2046609937

FollowLeadingVehicle:

python ego_vehicle_github.py FollowLeadingVehicle 20 1 0 5 12 0.15 -0.01 2000
python ego_vehicle_github.py FollowLeadingVehicle 20 1 0 6 34 0.8 -0.45 2000
python ego_vehicle_github.py FollowLeadingVehicle 20 1 0 6 14 0.9 -0.05 2000

If any further information needed, please let me know.
Thanks.

Blyron, 2024-04-17:
This should be fixed in the latest dev, branch

closed 2024-04-17 by Blyron