carla-simulator/carla #7409: Shader bug / weird artifacts / glow

alexx-km, 2024-04-11:
CARLA version: 0.9.15
Platform/OS: Windows / Ubuntu
Problem you have experienced: Sometimes, after loading a map and spawning vehicles / sensors, images saved by the RGB sensor have weird artifacts in them. This can be fixed by restarting CARLA.
What you expected to happen: -
Steps to reproduce: Happens randomly 
Other information (documentation you consulted, workarounds you tried): Current workaround is to try to detect those images using a histogram and restarting CARLA

[image]

[image]
...and after one frame ...

[image]

GoodarzMehr, 2024-04-11:
In my experience this usually happens when you max out your VRAM.

Blyron, 2024-04-12:
Yeah. When using Vulkan if you ran out of VRAM it usually happens. We suggest to run the server with -nullrhi command to consume less VRAM

GoodarzMehr, 2024-04-12:
Wouldn't -nullrhi disable the camera output altogether since there is no rendering?

Blyron, 2024-04-12:
It would be only for spectator camera, rgb sensors would work fine

GoodarzMehr, 2024-04-12:
Oh, I see. Is there any difference between that and -RenderOffScreen, or do they both do the same thing?

Blyron, 2024-04-13:
-RenderOffScreen still makes work GPU because Pixelstreaming is still working. Nullrhi doesn't

GoodarzMehr, 2024-04-14:
So I tried using -nullrhi instead of -RenderOffScreen but CARLA seems to segfault when spawning a camera. Specifically, I ran
`/home/carla/CarlaUE4.sh -nullrhi --carla-rpc-port=15418 -quality-level=Epic`, then connected a client and spawned a vehicle, but when I tried to spawn a camera attached to it I got this error:
```bash
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Segmentation fault (core dumped)
terminate called after throwing an instance of 'carla::client::TimeoutException'
  what():  time-out of 80000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:15418
[goodi:20728] *** Process received signal ***
[goodi:20728] Signal: Aborted (6)
[goodi:20728] Signal code:  (-6)
[goodi:20728] [ 0] /usr/lib/x86_64-linux-gnu/libc.so.6(+0x43090)[0x7efbc415f090]
[goodi:20728] [ 1] /usr/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb)[0x7efbc415f00b]
[goodi:20728] [ 2] /usr/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b)[0x7efbc413e859]
[goodi:20728] [ 3] /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x9e8d1)[0x7efbc2e078d1]
[goodi:20728] [ 4] /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa37c)[0x7efbc2e1337c]
[goodi:20728] [ 5] /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa3e7)[0x7efbc2e133e7]
[goodi:20728] [ 6] /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa699)[0x7efbc2e13699]
[goodi:20728] [ 7] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(+0x8ed1d5)[0x7efab81771d5]
[goodi:20728] [ 8] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(_ZN5carla6client6detail6Client5Pimpl7RawCallIJRjRKbEEEDaRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEDpOT_+0x8d)[0x7efab8191b7d]
[goodi:20728] [ 9] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(_ZN5carla6client6detail6Client5Pimpl11CallAndWaitIvJRjRKbEEEDaRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEDpOT0_+0x30)[0x7efab8173e60]
[goodi:20728] [10] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(_ZN5carla6client6detail6Client23SetActorSimulatePhysicsEjb+0x8d)[0x7efab816cb8d]
[goodi:20728] [11] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(_ZN5carla6client5Actor18SetSimulatePhysicsEb+0x3f)[0x7efab813a9bf]
[goodi:20728] [12] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(_ZN5carla15traffic_manager4ALSM10UpdateDataEbRKN5boost10shared_ptrINS_6client5ActorEEEbf+0x40c)[0x7efab82cd30c]
[goodi:20728] [13] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(_ZN5carla15traffic_manager4ALSM26UpdateRegisteredActorsDataEbRSt4pairIjdE+0x1bc)[0x7efab82cbc4c]
[goodi:20728] [14] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(_ZN5carla15traffic_manager4ALSM6UpdateEv+0x570)[0x7efab82ca6d0]
[goodi:20728] [15] /home/pf/.cache/Python-Eggs/carla-0.9.15-py3.7-linux-x86_64.egg-tmp/carla/libcarla.cpython-37m-x86_64-linux-gnu.so(_ZN5carla15traffic_manager19TrafficManagerLocal3RunEv+0x589)[0x7efab8287da9]
[goodi:20728] [16] /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xd6df4)[0x7efbc2e3fdf4]
[goodi:20728] [17] /usr/lib/x86_64-linux-gnu/libpthread.so.0(+0x8609)[0x7efbc4101609]
[goodi:20728] [18] /usr/lib/x86_64-linux-gnu/libc.so.6(clone+0x43)[0x7efbc423b133]
[goodi:20728] *** End of error message ***
Aborted (core dumped)
```
Am I missing something? If I need to launch a secondary server and connect to that, wouldn't that consume the same amount of VRAM in the end?

alexx-km, 2024-04-15:
Thank you for your quick reply! I tested the same script using a lower resolution for the cameras and now the bug seems to appear less often. However, it is not completely fixed and when it occurred, there was still 1.5GB of VRAM free...

Blyron, 2024-04-19:
Yeah... I missed that point... Nullrhi is used for Multigpu carla feature, that was my bad, try with RenderOffScreen, by the way, which are you NVIDIA DRivers version and libvulkan version?

closed 2024-06-28 by Blyron