carla-simulator/carla #7413: sensor.camera always cause memory leak since version 0.9.14 #7156

LuisPoveda, 2024-04-12:


Blyron, 2024-06-07:
Every sensor listen causes a memory leak we need to research about it

glopezdiest, 2024-06-10:
One comment about this, which might but be related. As far as I know, the `save_to_disk` function, which tends to be used in the listen callback, can create a memory leak. This is because it is a very slow function that stores the images in memory before saving them to disk, so if the simulation keeps ticking, more images keep piling up, eventually running out of memory. Also, the function doesn't wait for the image to be saved to disk before completing, which doesn't help when trying to synchronize it.

Blyron, 2024-07-08:
https://docs.google.com/document/d/1elDrbtYjHyMpLSzUMtd3MUz0G5fPyULQGz9TGWJb6kE/edit

closed 2024-09-20 by MarcelPiNacy-CVC