carla-simulator/carla #7420: Carla Content.tar.gz download URL changed

PatrickPromitzer, 2024-04-12:
Carla Version 0.9.15 source (and probably every version below)

If you get the following error
```
...
HTTP request sent, awaiting response... 404 Not Found
ERROR 404: Not Found.
```
, edit the "Update.sh" file.

Line 50
Old: 
```CONTENT_LINK=http://carla-assets.s3.amazonaws.com/${CONTENT_ID}.tar.gz```
New:
```CONTENT_LINK=https://carla-assets.s3.us-east-005.backblazeb2.com/${CONTENT_ID}.tar.gz```

Aperiss, 2024-04-19:
Thanks for the information, we will add this to the documentation.

closed 2024-04-19 by Aperiss

razr, 2024-12-04:
it is different now. I used it to build a `scenario_runner` container.

```bash
wget https://tiny.carla.org/carla-0-9-15-linux
--2024-12-04 12:08:23--  https://tiny.carla.org/carla-0-9-15-linux
Resolving tiny.carla.org (tiny.carla.org)... 66.220.23.67
Connecting to tiny.carla.org (tiny.carla.org)|66.220.23.67|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://carla-releases.b-cdn.net/Linux/CARLA_0.9.15.tar.gz [following]
--2024-12-04 12:08:24--  https://carla-releases.b-cdn.net/Linux/CARLA_0.9.15.tar.gz
```

fontray, 2025-01-09:
I use ubuntu 20.04 build carla 0.9.10 version, and my `make PythonAPI`  sucess, and i try to `make launch` it show this error

[image]

I change `Update.sh` file Line 50 to 
`CONTENT_LINK=https://carla-assets.s3.us-east-005.backblazeb2.com/${CONTENT_ID}.tar.gz`
can anyone tell me how to fix it ?
and I have another problem with UnrealEngine 4.24, this patch address doesn't work
`wget https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/UE_Patch/430667-13636743-patch.txt 430667-13636743-patch.txt`, is this cause the problem show in picture below 
[image]

PatrickPromitzer, 2025-01-13:
@fontray 
Make sure you are using the Carla Unreal Engine, and not the official one.
https://carla.readthedocs.io/en/0.9.15/build_linux/#unreal-engine

If you can't get it working with the Carla Unreal Engine, try using a newer Carla version.

garong, 2025-03-27:
@fontray Can you tell me the correct URL? I need to use version 0.9.10, but I'm stuck at the UnrealEngine Patch part of the Linux Build.

fontray, 2025-03-27:
@garong  
```
 wget https://carla-releases.s3.us-east-005.backblazeb2.com/Linux/UE_Patch/430667-13636743-patch.txt 430667-13636743-patch.txt
 patch --strip=4 < 430667-13636743-patch.txt