carla-simulator/carla #7423: Unexpected values on agents' acceleration got by agent.get_acceleration()

BoluoMonster, 2024-04-13:
I am working on getting the ego-agent's acceleration by agent.get_acceleration(). However, I find the resulting values are unrealistic. Here is the data log:
[image]
All values in the image are with the unit $m/s^2$. The throttle is constantly 0.75, but the acceleration changes sharply, which is weird.
By the way, the simulation is in synchronous mode.

glopezdiest, 2024-05-10:
Could you provide more information about the exact experiment you are running? Are you going in a straight line or turning? which vehicle is it?

The only thing that I can say right now is that constant throttle doesn't mean constant acceleration. Current speed, aerodynamics, gears, motors... all these factors affect the vehicle differently, hence why I need more information about the experiment

BoluoMonster, 2024-05-11:
@glopezdiest Thanks for replying! Here is the experimental setting:
Vehicle: vehicle.audi.a2
Controller: PID controller from "PythonAPI"
Testing scenario: a bend in Town02
[image]
The vehicle is tracking a lookahead point that is 2.0m ahead.
[image]
Following is what will be observed when the vehicle is turning.
[image]
We can find that the acceleration changes sharply. And it also happens when the car is moving straight forward.
[image]

YuhengZhi, 2024-08-26:
TL;DR This is expected. The throttle value is not a one-to-one mapping to physical acceleration, especially if you are making a turn. Many other factors like gear ratio, final ratio, tyre stiffness, wheel radius, current engine RPM, tyre/road friction, engine damping, engine torque curve, steering angle, are involved to in the calculation of physical acceleration.

The "throttle" value between [0,1] actually should not be considered an physical acceleration value, but more a scalar ratio of how much torque the engine is outputting from the max torque it can output at the current RPM level. [These lines](https://github.com/CarlaUnreal/UnrealEngine/blob/e8c73304f9f3f91ba12c997516deeb2434e4489e/Engine/Source/ThirdParty/PhysX3/PhysX_3.4/Source/PhysXVehicle/src/PxVehicleUpdate.cpp#L1424-L1428) from PhysX 3.4 used by UnrealEngine gives a peek into how the engine torque is actually determined by throttle:
```
PX_FORCE_INLINE PxF32 computeEngineDriveTorque(const PxVehicleEngineData& engineData, const PxF32 omega, const PxF32 accel)
{
	const PxF32 engineDriveTorque=accel*engineData.mPeakTorque*engineData.mTorqueCurve.getYVal(omega*engineData.getRecipMaxOmega());
	return engineDriveTorque;
}
```
The `accel` value is a smoothed version of the user input throttle and is also between [0, 1]. The engine torque as you can see is determined by the current RPM of the engine (omega, just under a different unit), the "torque curve", and the scalar throttle/accel.

The engine torque will go through gear ratio and the final ratio, the radius of the car's wheels and stiffness of the tyre to produce a relative "slip" on the road surface, which in return generates a friction force to push the vehicle forward. What makes this more complicated is "slip" does not only happen on the longitudinal direction but also lateral, which provides turning force. An actual slip can also happen on both directions if the friction value of the wheel or the road surface is small. All of these dynamics are wrapped up into an implicit function that requires solving a set of dynamic equations and constraints. 

Refs:
1. [PxVehicleUpdate.cpp](https://github.com/CarlaUnreal/UnrealEngine/blob/e8c73304f9f3f91ba12c997516deeb2434e4489e/Engine/Source/ThirdParty/PhysX3/PhysX_3.4/Source/PhysXVehicle/src/PxVehicleUpdate.cpp) from PhysX_3.4 (requires linking your github account to unreal as in [this guide](https://www.unrealengine.com/en-US/ue-on-github))
2. https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Vehicles.html
3. https://forums.developer.nvidia.com/t/nature-of-damping-rate-and-torque-curve-and-relationship-with-speed/163744

BoluoMonster, 2024-08-27:
Thanks for your reply! I got it :)

closed 2024-08-27 by BoluoMonster