carla-simulator/carla #7425: Use PythonAPI to change the color of my imported vehicles

zhoujiawei3, 2024-04-13:
I need to use PythonAPI to change the color of my imported vehicles
In Carla, I can use
    ego_vehicle_bp_origin = blueprint_library.find('vehicle.audi.etron')
    ego_vehicle_bp_origin.set_attribute('color', color_name)
It is possible to use set_attribute to change the color of a vehicle when loading an officially provided vehicle by Carla.
But if I use my own newly created vehicle to import into Carla, and then use ego_vehicle_bp_originblueprint_library.find('vehicle.mycar.mycar.')
ego_vehicle_bp_origin.set_attribute('color', color_name)
Such code will report error : IndexError: attribute 'color' not found.

gsmario, 2024-04-14:
You can add colors by going in to the BP_VehicleFactory and looking for your vehicle there.
Expand the information of the vehicle and add colors of your liking, if you want to see a preview, you can drag your asset in to the scene (without starting playing) and modifing the color in real time, then copy the color code and paste it in the vehiclefactory.

joel-mb, 2024-05-10:
Closing this issue as is already answered and no feedback from the user is provided.

closed 2024-05-10 by joel-mb