carla-simulator/carla #7457: Added new pedestrian, cant spawn Actor

MTeuberItemis, 2024-04-17:
Im using carla 0.9.13
built it on Windows from Source
Python 3.7.9

I tried adding a new pedestrian, a wheelchair user.

To realize that, I added the wheelchair first as a skeletal mesh in static
```
Content/
├─ Carla/
│  ├─ Static/
│  │  ├─ Pedestrian/
│  │  │  ├─ WheelchairUser_/
│  │  │  │  ├─ Meshes_/
│  │  │  │  │  ├─ Phsys_Wheelchair
│  │  │  │  │  ├─ SK_Wheelchair_Skeleton
│  │  │  │  │  ├─ SM_Wheelchair
│  │  │  │  ├─ MI/
```

Afterwards I created an Animation Blueprint that sits down the Generan skeletal mesh:
Content\Carla\Static\Pedestrian\Animations\WheelchairUser\ABP_WheelchairUser

Then I created a general BP for wheelchair users that is the BP_Walker with the wheelchair sm added and some additional functionalities, such as sitting down the pedestrian. This is inspired by the way 2-wheeled vehicles handle sitting. 
Content\Carla\Blueprints\Walkers\BP_WheelchairUser

As a child of that I created BP_WheelchairUser_v1, which is data-only and assigns an ABP and Skeletal Mesh. 
I added this BP to the WalkerFactory and gave it a generation value of 3, so I can call it easier

Now when I run a modified generate_traffic.py, that only spawns BPs with generation value 3 (so only wheelchair users). 
Unfortunately this leads to following error:

`ERROR: Unknown error while trying to spawn actor`

in the UE Output log this looks as follows:

`LogCarla: Spawning actor 'walker.pedestrian.0050'
LogCarla: Warning: Failed to spawn actor 'walker.pedestrian.0050'
LogCarla: Error: Actor not Spawned
LogCarlaServer: Responding error: Unknown error while trying to spawn actor`

Im guessing this is an error with the way I set up the new pedestrian. But I cant find any documentation about importing own walkers. Has anyone achieved this and can help me?

MTeuberItemis, 2024-04-17:
I noticed some more weird behaviour. I noticed that the BPs actually get spawned in the scene, although the error message is shown. 
However they are not moving and are having the human model in wrong spot, while my created blueprint is posed correctly (when I just place it in the scene). 

pose when spawned by script (with error message):
[image]

pose when placed in scene:
[image]

DnDare, 2024-04-25:
Hi, I'm not sure if I understood your issue.
The reason for the ABP is to detect which state the pedestrian is (idle, turning, walking running), and them play the according animation. 
This is made at the EventGraph
[image]

DnDare, 2024-04-25:
Then at the Animgraph tab, is where the State Machine handles the animations.
[image]
Inside there is a Blendspace (BS_GEN3_v2), handles idle, walking, running animations, 
The  right and left turn animations are handled outside of the Blendspace.

@MTeuberItemis   have you changed the animations at the AnimGraph and StateMachine?

Else , if I the issue is different , then tell me.

DnDare, 2024-04-25:
The approach you reviewed from the 2-wheels vehicles, uses a different method. Uses Blueprints to place the skeletal mesh in position, then the hands are constrained to the handlebar, and the feet have constrains too.
--------
I've to review it, but if you can make some animations set I'd go for the Blendspace/StatMachine option.
Tell me if you want to try this option, or if getting some animations set might be a blocker.

MTeuberItemis, 2024-04-29:
I set up the ABP by copying the nodes from the 2WheelerABP and made the constraints work again. I for now just wanted to sit down the wheelchair user, without any additional animation. The sitting-downp-art of the ABP was working and I still dont really know, why they didnt move. 
But I actually found a solution in the last days. Before I duplicated the BP_Walker and created a separate BP for all wheelchair bps as a base. Now I tried to modify the BP_Walker and give him functions that attach a wheelchair and sit him down based on a bool. I still had to adjust the ABP but it seems like carla is only excepting the original BP_Walker as a base for modified pedestrians

DnDare, 2024-05-10:
I couldn't review so much on this issue, 
The riders are constarined to a bone that is placed at the bike's seat.

Have you constarined the pedestrian to a joint located at the seat?

And by the image you show, I don't think the arm constraints are working at the simulation. 
I may be wrong, but maybe constraints are not working properlly

DnDare, 2024-05-10:
[image]
At the editor, can you see your bp broken like this one? 
They're not really broken, it is because of the constraints.  This way you can tell they're working.
Hope this helps.

MTeuberItemis, 2024-05-16:
I managed to fix it now, by not using constraints, but just an animationBP that is sits the character in the right position. Its a bit more constrained to that one type of wheelchair, but works for now.

closed 2024-06-07 by DnDare