carla-simulator/carla #7473: Generate more python *.whl in the package

xavisolesoft, 2024-04-19:


PatrickPromitzer, 2024-04-24:
It is only a guess, because I edited some files and got a python 3.8 *.whl

https://github.com/carla-simulator/carla/blob/dev/Util/BuildTools/BuildPythonAPI.sh
line 19
```
PY_VERSION_LIST=3
```

The same here
https://github.com/carla-simulator/carla/blob/dev/Util/BuildTools/Setup.sh
line 15
```
PY_VERSION_LIST=3
```

On Ubuntu, if you can change that line to
```
PY_VERSION_LIST=3.8
```

It was a fix for Ubuntu 22.04, until Ubuntu 22.04 stopped to support clang-7

MarcelPiNacy, 2024-05-03:
> It is only a guess, because I edited some files and got a python 3.8 *.whl
> 
> https://github.com/carla-simulator/carla/blob/dev/Util/BuildTools/BuildPythonAPI.sh line 19
> 
> ```
> PY_VERSION_LIST=3
> ```
> 
> The same here https://github.com/carla-simulator/carla/blob/dev/Util/BuildTools/Setup.sh line 15
> 
> ```
> PY_VERSION_LIST=3
> ```
> 
> On Ubuntu, if you can change that line to
> 
> ```
> PY_VERSION_LIST=3.8
> ```
> 
> It was a fix for Ubuntu 22.04, until Ubuntu 22.04 stopped to support clang-7 (maybe it was clang-8)

Hi Patrick, this is actually for the project roadmap. So this may be in the next release, which should change the build system to just CMake (so the BuildTools folder will be removed most likely).