carla-simulator/carla #7478: 'C:\Program" is not recognized as internal or external command error when running "make PythonAPI"

ymoghadd, 2024-04-19:
Hi,

When I run "make PythonAPI" I get the error below. If you anyone has a solution to this issue please let me know.

  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: 'C:/Users/yasi4/carla/Build/zlib-source/build/CMakeFiles/CMakeScratch/TryCompile-cjtqnx'

    Run Build Command(s): "C:/Program Files/CMake/bin/cmake.exe" -E env VERBOSE=1 nmake -f Makefile /nologo cmTC_cfbf2\fast
        C:\Program Files (x86)\GnuWin32\bin  -f CMakeFiles\cmTC_cfbf2.dir\build.make /nologo -L                  CMakeFiles\cmTC_cfbf2.dir\build
    'C:\Program' is not recognized as an internal or external command,
    operable program or batch file.
    NMAKE : fatal error U1077: 'C:\Program' : return code '0x1'
    Stop.

Blyron, 2024-04-20:
Are you using x64 Native Tool Command Prompt to run "make PythonAPI"?

ymoghadd, 2024-04-21:
Yes, specifically for VS 2019

BaoJie456, 2024-05-29:
Same problem, have you solve this?

> Yes, specifically for VS 2019

vkortvelyesi, 2024-10-26:
Hi, I had the same issue on Windows 11. Try this solution:

https://github.com/carla-simulator/carla/discussions/5351

It worked for me.

closed 2024-11-08 by Blyron