carla-simulator/carla #7480: Get_actors() problem

xinyhu888, 2024-04-21:
Kindly Asking
CARLA version:0.9.15
Platform/OS:Windows
Problem you have experienced: I have an custom actor which is not spawned by the world.spwan_actor() method in PythonAPI, but I want to get it, like world.get_actors() method does, BUT i cant do it now, there are some problem. BTW, I have added it to the Vehicle Factory structure.
What you expected to happen: Would you please offer some ideas about how can I solve this problem.

MarcelPiNacy, 2024-04-24:
Hi, how are you currently spawning it? Worst case, you could directly expose a method to the Python API to retrieve the information you want out of it.

PatrickPromitzer, 2024-04-24:
I don't know how you code looks like, but I can give you advice to isolate the problem.

1.) If you look at the blueprint list, can you find the actor you added?
```python
bp_lib  = world.get_blueprint_library()
actor_type_ids = [str(bp_item.id) for bp_item in bp_lib]
for type_id in actor_type_ids:
    print(type_id)
```
2.) Does carla output an error if you spawn the actor? (python API or server)
3.) Is an object at the position you want to spawn the actor? (floor, building, plant, ...)
Try to spawn it with z=50 to have free room
4.) If you spawn the actor, can you see the position and size?
```python
actor = world.spawn_actor(actor_bp, actor_transform)
bounding_box_size = [
    actor.bounding_box.extent.x * 2,
    actor.bounding_box.extent.y * 2,
    actor.bounding_box.extent.z * 2,
]
for size_item in bounding_box_size:
    print(size_item)
```

It helps if you add the error text if you get one, and the code you are using.
It makes it easier to find the problem.

Blyron, 2024-10-18:
Editor spawned actors are not exposed to the pythonAPI sorry for that, you will need to add your actor to a factory or create one on your own. Sorry for that

closed 2024-10-18 by Blyron