carla-simulator/carla #7481: Runtime error while running generate_traffic.py

risbansal, 2024-04-21:
CARLA version: 9.15
Platform/OS: Ubuntu 22.04
Problem you have experienced: RuntimeError: rpc::rpc_error during call in function apply_batch
What you expected to happen: To see vehicles,pedestrians etc added to the road.
Steps to reproduce: I ran generate_traffic.py
Other information (documentation you consulted, workarounds you tried):

I have followed the carla documentation. I was able to run this once. But the error started showing up when I tried the 2nd time. There was a similar question asked in 2019. The suggestion was to work with Python 3.7, and I am already using Python 3.7.

I would really appreciate if someone can help me understand this error and how to solve it.

eindino1, 2024-04-22:
Hi,
Got the same issue but i'm using:
CARLA version: 9.15
Platform/OS : Windows 10

I can run dynamic_weather without any problem but all exemple who use car have different issue link to rpc.

risbansal, 2024-04-22:
Yes same here. I can also run dynamic_weather

risbansal, 2024-04-22:
I just solved that issue. I was using Python 3.7.16 earlier and was facing the issue. I read somewhere that Carla works best with Python 3.7.0. So I setup a new environment in anaconda with python 3.7.0 and then did make PythonAPI and make launch again. Hope this solves your issue to @eindino1.

eindino1, 2024-04-23:
Unfortunately, it didn't work for me. I've discovered that it may be a make problem, as I can't rebuild, so I'm guessing that the PythonAPI make didn't work properly.

risbansal, 2024-04-23:
Update: The error came back for me.
I was runnning a code with Keras and Tensorflow today. I had to down grade Keras and tensorflow since I am using python 3.7 and then I ran into the same error. And it is always accompanied with the following warning:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.15 
WARNING: Simulator API version  = 0.9.15-169-g063cc9d90 

Can an expert please help us please

eindino1, 2024-04-24:
I found the error on my side, I tried to execute the code in another folder but line 17 of generate_traffic.py needs the path to the .egg and it's not a os path, it's a hard-coded path.
```
 sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
```

change the path to an os path solution the issue

risbansal, 2024-04-24:
ok I see. That was not an issue for me as I am using ubuntu. I also realized that on my end the error comes back whenever I do pip install carla. For some reason this creates and inconsistency in version for client and simulator. 
Also import carla works with having to pip install it as long as I use Python 3.7 
But this is creating a problem because I am not able to use later versions of Keras and tensorflow and cannot use some models that are part of the higher version. I am stuck with old versions because of Python 3.7. Does anyone have a solution to this?

ahaider1011, 2024-04-25:
I think I am having a similar issue
#7496

joel-mb, 2024-04-25:
Hi @risbansal.
CARLA currently supports from python `3.7` to `3.10`.
If you are building from source, you can choose the desired python version in the following way:
```bash
make PythonAPI ARGS="--python-version=3.10"
```
The version mismatch indicates that you are using a simulator and a client that were built in different commits. This can produce errors as the code is changing rapidly. Normally this can happen if you have previously installed a CARLA client using `pip` (this will install the latest CARLA client available in `pypi` https://pypi.org/project/carla/).

closed 2024-04-26 by joel-mb

risbansal, 2024-04-27:
I understand this a lot better now. Thanks a lot for the explaination