carla-simulator/carla #7482: Installation and Execution Errors with Carla on WSL

eyabesbes, 2024-04-22:
Hi,
after following Carla's documentation for installation, I encountered the following errors while executing 
`` ./CarlaUE4.sh: `` on WSL:

``
chmod: changing permissions of '/content/carla/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping': Operation not permitted
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps. ``

I attempted to resolve this issue by running:
`` unset SDL_VIDEODRIVER
./CarlaUE4.sh -vulkan -RenderOffscreen -nosound `` 

I also tried using the [xorg-less / headless]
(https://carla.readthedocs.io/en/0.9.12/adv_rendering_options/#setting-off-screen-mode-version-0912) with `` ./CarlaUE4.sh -vulkan -RenderOffscreen ``

joel-mb, 2024-04-25:
Hi @eyabesbes 
Could you explain your issue in more detail?
It seems you have permissions issues when making `CarlaUE4-Linux-Shipping` executable. But the server seems to be running.

joel-mb, 2024-05-06:
Closing the issue. Please, feel free to reopen the issue if needed.

closed 2024-05-06 by joel-mb