carla-simulator/carla #7489: static.prop.mesh does not attach to parent

saippua, 2024-04-23:
**CARLA version:** 063cc9d902f09aa37e789330f590d94c53b66d31 (currently latest on dev branch)
**Platform/OS:** Windows 10
**Problem you have experienced:** 
`static.prop.mesh` does not attach to parent actor like other static prop actors do.
[image]

**What you expected to happen:**
When I spawn a `static.prop.mesh` with `world.spawn_actor(bp, tf, attach_to=PARENT)` the `static.prop.mesh` should attach to the parent. This is how it works with other props, such as `static.prop.haybale` or `static.prop.colacan`

Instead, what happens is the `static.prop.mesh` is physicalized, so it falls down with gravity, and movement of the parent actor does not transfer to the `static.prop.mesh`

**Steps to reproduce:**
Here is a minimum example python file. It spawns a vehicle and two actors attached to it. One is a `static.prop.haybale` and the other is `static.prop.mesh` with haybale model and mass of 1. The `static.prop.mesh` falls down when it spawns and does not follow the car, while the `static.prop.haybale` does.
[transform_test.py](https://github.com/carla-simulator/carla/files/15075185/transform_test.txt)

MarcelPiNacy, 2024-04-24:
Hi,

It looks like physics simulation is disabled by default for meshes whose mass is set to 0. It may be that the haybale's mass defaults to this value.
I will come back to this later with more info.

Marcel

saippua, 2024-04-25:
I've tested with mass=0 for `static.prop.mesh`, and it still doesn't behave the same as the other `static.prop.*` blueprints.
The `static.prop.mesh` is not attached to the parent actor (it does not move when the parent moves) and it is also spawned w.r.t the world coordinate system, not the parent actor coordinate system like it should.

[image]

Here is an example. Both static props are spawned with a `world.spawn_actor(bp, tf, attach_to=TESLA_CAR)` where `tf` is a transform of `(0,0,5)`, no rotation (they both have a 1m Y offset so they don't spawn overlapping)

The `static.prop.mesh` spawns above the world origin, and the `static.prop.haybale` spawns over the parent origin. Also, as the car spawns slightly above the ground, it falls down at the beginning of the simulation and I can see that the `static.prop.haybale` is rigidly attached to the car, whereas the `static.prop.mesh` is not.

anrp-tri, 2025-09-30:
Drive by comment: looks like this might be related to the static.prop.mesh factory explicitly setting the object to be not movable (in unreal terms): 
UE4: https://github.com/carla-simulator/carla/blob/6fe2ffa3880c0ebd91593d6957927673e1736d82/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Actor/StaticMeshFactory.cpp#L72-L84
UE5: https://github.com/carla-simulator/carla/blob/779675543bc30cd974604c56c56b76d2cbf71b66/Unreal/CarlaUnreal/Plugins/Carla/Source/Carla/Actor/StaticMeshFactory.cpp#L75-L87