carla-simulator/carla #7496: RuntimeError: rpc::rpc_error during call in function spawn_actor_with_parent

ahaider1011, 2024-04-24:
CARLA version: 0.9.15
Platform/OS: Linux Ubuntu 22.04 LTS
Problem you have experienced:

I have built Carla from source using git clone and it works properly. Just when I attach a sensor, Camera, Lidar etc. it gives this Runtime Error. Is this a Carla issue?

Steps to reproduce:
Build latest commit on linux. 
Make PythonAPI ( tried with python 3.7.0, 3.7.16, 3.8 and 3.9) same result for all
Then make launch
Unreal engine starts 
If run any script with just spawning of vehicle > the vehicle spawns
But if attach any sensor for example a lidar or camera. The script fails with the runtime error of RPC function spawn_actor_with_parent
and if run any other examples, gives same error

ahaider1011, 2024-04-25:
@MarcelPiNacy 
[image]

joel-mb, 2024-04-25:
@ahaider1011 The version mismatch indicates that you are using different versions for the CARLA simulator and the Client API.

First make sure you uninstall any previous CARLA client:
```
python3.9 -m pip uninstall carla
```

Then run the `manual_control.py` script again, making sure you use the same python version you used for building:
```
python3.9 manual_control.py
```

The version mismatch should disappear at this point.

joel-mb, 2024-05-07:
Closing the issue. Please, feel free to reopen it if needed.

closed 2024-05-07 by joel-mb

zolanElleyoneD, 2024-06-13:
I have the same error using ROS-bridge for ROS1 on version 0.9.15 on Ubuntu 20.04 (and python 3.8). I can't load any sensor and the simulation seems to be accelerated. 
[image]
I tried the solutions explained in the FAQ (https://carla.readthedocs.io/en/0.9.10/build_faq/) but it does not work. I also tried using a different environment with conda for the client. I also tried removing the Carla lib installed with pip to just let the egg file (using export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla), still does not work. I have both package and built 0.9.15 and get the same error, however it works well with 0.9.13 on the same machine.