carla-simulator/carla #7505: load external xodr cause segmentation fault

SdKay, 2024-04-26:
- CARLA version: 0.9.13
- Platform/OS: ubuntu20.04
- Problem you have experienced: **xodr load segmentation fault**
- What you expected to happen: xodr load success
- Steps to reproduce:
  - test code:
```python
import sys
import os
import carla

client = carla.Client('localhost', 2000)
client.set_timeout(10.0)

with open('town01.xodr', encoding='utf-8') as od_file:
#with open('faild.xodr', encoding='utf-8') as od_file:
    try:
        data = od_file.read()
    except OSError:
        print('file could not be readed.')
        sys.exit()

vertex_distance = 2.0  
max_road_length = 500.0 
wall_height = 1.0       
extra_width = 0.6      
world = client.generate_opendrive_world(
    data, 
    carla.OpendriveGenerationParameters(
        vertex_distance=vertex_distance,
        max_road_length=max_road_length,
        wall_height=wall_height,
        additional_width=extra_width,
        smooth_junctions=True,
        enable_mesh_visibility=True
    )
)

tm = client.get_trafficmanager(8000)   # segmentation fault here!!
```
`town01.xodr` can be generate world and call `client.get_trafficmanager` normally, but `faild.xodr` cracked when call `client.get_trafficmanager`

[xodr.zip](https://github.com/carla-simulator/carla/files/15124661/xodr.zip)

Blyron, 2024-04-26:
Could you try this with 0.9.15 or latest dev branch?

PatrickPromitzer, 2024-04-26:
It is hard to check an openDRIVE file without a tool, because the file can get big fast.
I normally check with https://odrviewer.io/ if the file can be loaded.

I let it run through a tool I wrote to check if the roads are right.
I believe the faild.xodr file doesn't look right. (I could be wrong)
[image]
[image]

Blyron, 2024-04-26:
I mean to run your script on latest version of CARLA

SdKay, 2024-04-28:
> Could you try this with 0.9.15 or latest dev branch?

Not yet, because we has develop lots of features based on 0.9.13 docker....Maybe I can test it on 0.9.15

SdKay, 2024-04-28:
> It is hard to check an openDRIVE file without a tool, because the file can get big fast. I normally check with https://odrviewer.io/ if the file can be loaded.
>
> I let it run through a tool I wrote [...]

I also think there is a problem with xodr. I have also used this website to load it, but it is always in the loading state.

PatrickPromitzer, 2024-04-29:
With the first post you told that town01.xodr can be generated, and faild.xodr can't.
With the website not be able to load it, and seeing the output of the tool I wrote, I believe the faild.xodr file it too broken to work in Carla.
With how the roads look, I would try to  make a new *xodr file, because repairing could take a lot of time.

Blyron, 2024-04-29:
Yeah I agree with PatrickPromitzer

Blyron, 2024-05-03:
Have you been able to verify your xodr was correct?

SdKay, 2024-05-07:
> Have you been able to verify your xodr was correct?

Yes, there is something wrong with it, I'm trying to find out where it is and how it comes.

Blyron, 2024-05-10:
I am closing this issue as the error is in the xodr so feel free to open it if something comes up