carla-simulator/carla #7514: How to use any external sumo-Traci python code or Carla script(for sensor acquisition ) in SUMO-CARLA so-simulation

arscode, 2024-04-28:
# Issue: unable to plut custom sumo-traci and/or carla python script in the sumo-carla co-simulation
### CARLA version: 0.9.15
### Platform/OS: windows 11

## Expectation: to run sumo-carla co-simulation with sumo-traci custom code and carla script to get sensor data

I am trying to plug my own custom code written for sumo-traci simulation and carla sensor integration in the sumo-carla co-simulation. both working separately but I can't use them while using co-simulation using run_synchronization. I need help to understand where my codes needs to be plugged? do I have to put the codes in the sumo_simulation.py and carla_simulation.py files and use run_synchronization.py for co-simulation?

joel-mb, 2024-05-07:
Hi @arscode 
Cloud you explain your use case in more detail? In principle, changes to the source code are not needed.

You can find some references here: https://drive.google.com/file/d/10DgMNUBqKqWBrdiwBiAIT4DdR9ObCquI/view

arscode, 2024-05-10:
Hi @joel-mb 
I went through that lecture as well, but it seems I can run co-simulation only using run_synchronization.py. Now if I want to add some sensors in the ego vehicle or change car-following model in the SUMO then I need to plug my own code in the co-simulation. I found in the source code, run_synchronization.py it calling sumo_simulation, carla_simulation and bridge_helper but moreover all of these working on the clock synchronization, I'm not sure how can I put some external/ my own code in the co-simulation. I can separately use traci and carla API for sensor acquisitions or leader-follower implementation, etc. but I cant run them in co-simulation. I need to gather sensor data from CARLA-SUMO co-simulation and mange the TLS/car-following models using SUMO. Please let me know if I have conveyed my concern properly.