carla-simulator/carla #7516: make PythonAPI repeated error [/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang.]

MOHAMMEDHUSSEIN94, 2024-04-29:
CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04
Problem you have experienced: when I run [make PythonAPI] I get [CMake Error at /usr/share/cmake-3.22/Modules/CMakeDetermineCCompiler.cmake:49 (message):
  Could not find compiler set in environment variable CC:

  
  /Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/home/mohammed/carla/Build/rpclib-v2.2.1_c5-c10-libcxx-build/CMakeFiles/CMakeOutput.log".
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1
] I added [export UE4_ROOT=~/UnrealEngineCarla] at the end of bashrc and then ran source and used echo $UE4_ROOT to make sure the directory is that of unreal engine and still getting the same error
What you expected to happen: build to succeed
Steps to reproduce: do not know
Other information (documentation I consulted, workarounds you tried): tried all documentations , github, stackoverflow, gemini and chat gpt 4

Blyron, 2024-04-29:
Please, do a make clean a retry following all steps in the guide
https://carla.readthedocs.io/en/latest/build_linux/

Blyron, 2024-04-30:
it seems you haven't installed libc++

MOHAMMEDHUSSEIN94, 2024-04-30:
> libc++

no it is installed and I am getting the same error. 
[image]

by the way I am using the latest unreal engine named carla(default) 4.26.2 cloned from your repository and set the path of UE4_ROOT to be its directory through bashrc and sourced bashrc

MarcelPiNacy, 2024-05-03:
Hi,

Could you check that the path reference in CC points to a valid clang compiler?
That is make sure that `/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang` really exists.

Marcel

loveskycloud, 2024-06-13:
I have encountered the same problem. Has it been resolved later?

rui-shen-afk, 2024-06-25:
Is there an update on this issue?

ZiQ-Li, 2025-02-25:
I have met the same problem, any other solution?

ZiQ-Li, 2025-02-25:
@rui-shen-afk @loveskycloud

qzhq747, 2025-06-28:
I face the same issue, and solve it by setting the correct UE4_ROOT.
export the right UE4_ROOT in the terminal or edit it in the .bashrc.

kanbarashen, 2025-09-16:
Same problem. "/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang" is in UE4 so the problem is Setup.sh cannot find the right UE4_ROOT
I set UE4_ROOT in the terminal but it not works so I set it in Setup.sh
try add
`export UE4_ROOT= the/path/to/your/UE4`
after
`source $(dirname "$0")/Environment.sh `
before
`export CC="$UE4_ROOT/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang"`
which works for me

User13720204240, 2026-01-16:
> I face the same issue, and solve it by setting the correct UE4_ROOT.我也遇到同样的问题，通过设置正确的 UE4_ROOT 解决了。 export the right UE4_ROOT in the terminal or edit it in the .bashrc.在终端导出右 UE4_ROOT，或者在 .bashrc 中编辑。

How do you do this? Can you elaborate on the process you operate?