carla-simulator/carla #7528: Linux build patch is not available

NBRodrigues, 2024-04-29:
Hi, 

I'm trying to install Carla 0.9.15 on a Linux machine but the patch is not available on aws container

get https://carla-releases.s3.eu-west-3.amazonaws.com/Linux/UE_Patch/430667-13636743-patch.txt ~/430667-13636743-patch.txt
patch --strip=4 < ~/430667-13636743-patch.txt

Best regards
Nelson

Blyron, 2024-04-30:
We have moved away frome aws, we are using backblaze now 
https://carla-releases.s3.us-east-005.backblazeb2.com/Linux/UE_Patch/430667-13636743-patch.txt

But to build Carla 0.9.15 you have to use Unreal Engine fork

NBRodrigues, 2024-04-30:
So, the patch is for UE 4.24, the UE 4.26 don't need the patch.

In the readme.md the information stills there
https://github.com/carla-simulator/carla/tree/0.9.15

Thanks
Nelson

Blyron, 2024-04-30:
In the 0.9.15.2 We have updated the MD. Are you happy to close this issue?

closed 2024-04-30 by NBRodrigues