carla-simulator/carla #7539: Car Paint Shader

Madecu, 2024-05-02:
Enhance vehicles' paint shader to make them more photorealistic:

DONE: 
The car paint shader is working, it allowed us to make three different types of car paint. We want to add dirt to the car paint but it it is not finished yet, it will be on hold until it becomes a priority.

[image]
[image]
[image]

closed 2024-05-02 by Madecu

lt-wang, 2024-09-01:
How it was achieved

AinaRoca, 2024-09-06:
This is the car paint material, the rest is dirt, but what you can see on the images is pretty much this:
The most characteristic things are the clear coat roughness and the flakes.

[image]

As you can see, you have the roughness which is part of the paint and the clear coat roughness that looks like the final layer on top of the paint, playing with those two values will allow you to make a shiny car like the red one or a matte car like the silver one, and many inbetweens.

On the other side there are the flakes. I've used this texture as an alpha. I chose a slightly lighter color from the car paint for those flakes to make it look like it shines. But the color diference is not enough to make it "sparkling", so I use this normal texture to give it some volume, that way, the reflection will change, but at the same time, because of the clear coat roughness you will see that the car surface is smooth. And this will allow you to make a car similar to the orange one.

[image]

Hope this helps!

lt-wang, 2024-09-06:
> 这是汽车油漆材质，其余的是污垢，但您在图像上看到的基本上是这样的： 最具特色的是透明涂层的粗糙度和剥落。
> 
> ![图像](https://private-user-images.githubusercontent.com/119680845/365109432-a4648410-a430-404c-b865-8cc8d0ed0253.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MjI3NTgsIm5iZiI6MTcyNTYyMjQ1OCwicGF0aCI6Ii8xMTk2ODA4NDUvMzY1MTA5NDMyLWE0NjQ4NDEwLWE0MzAtNDA0Yy1iODY1LThjYzhkMGVkMDI1My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNlQxMTM0MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hZWI0NTcwZTdjNTdlY2I1MTMzMmZhMTgwODA2NzkxYTNmNTkwNDVhYTI3MjQ2ZjQwMDg2ZDA1ZDIzNzI4ZjRjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.MCuVLJ0uXl1UPNUhI8HuzvQFor8lZp5iYW2WZ4jLGWU)
> 
> 正如您所见，您拥有作为油漆一部分的粗糙度和看起来像油漆顶部的最后一层的透明涂层粗糙度，使用这两个值可以让您制作出像红色一样的闪亮汽车或像银色一样的哑光汽车，以及许多介于两者之间的颜色。
> 
> 另一边是薄片。我已将此纹理用作 alpha。我为这些薄片选择了比汽车油漆略浅的颜色，使其看起来像是闪闪发光。但颜色差异不足以使其“闪闪发光”，所以我使用这种普通纹理为其赋予一些体积，这样，反射就会发生变化，但同时，由于透明涂层的粗糙度，您会看到汽车表面很光滑。这将使您能够制作出类似于橙色的汽车。
> 
> ![图像](https://private-user-images.githubusercontent.com/119680845/365115854-a0c25450-9eb4-4360-9ff3-206da6f90aee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU2MjI3NTgsIm5iZiI6MTcyNTYyMjQ1OCwicGF0aCI6Ii8xMTk2ODA4NDUvMzY1MTE1ODU0LWEwYzI1NDUwLTllYjQtNDM2MC05ZmYzLTIwNmRhNmY5MGFlZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNlQxMTM0MThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNjQ4ZDI1Y2EyMDBiODFiNjUxYjZhZjNiYWQ3YWYzZjk2NTBjOGUzMDdkOTI2ZGMxZTc4OTMzZDdhZDJiYjVlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.MPNoQ2JJsZLiGDNRP91Z41dNVw9MvbmWwnU0snenIbs)
> 
> 希望这有帮助！

Thank you for your generous reply, it really helpes me a lot.