carla-simulator/carla #7577: Implementation of CARLA-Trained Models on Physical Vehicles

knowaiser, 2024-05-05:
Hello,

I am currently working on a project that involves training and validating autonomous driving models within the CARLA simulation environment. One crucial aspect of my research is the testing and validation of these models on physical vehicles.

As I delve into this phase of my project, I am faced with the challenge of finding an affordable and practical method to implement the testing using a physical vehicle. My objective is to seamlessly transition the models trained within CARLA to real-world scenarios for thorough evaluation.

I am reaching out to inquire if there are any recommendations or best practices for achieving this goal. Specifically, I am interested in exploring the feasibility of utilizing RC cars as proxies for real vehicles in this context.

Could you kindly provide insights into whether certain types of RC cars or other affordable alternatives would suffice for this purpose? 

Thank you very much in advance.

PabloVD, 2024-05-10:
We can suggest you to look at some research works like [this one](https://arxiv.org/pdf/1710.02410) or [this repo](https://github.com/JahodaPaul/autonomous-car-chase), where models trained in CARLA simulations have been implemented in real toy physical vehicles, similar to your approach.

knowaiser, 2024-05-12:
> Indeed, applying models trained in simulations to physical vehicles is one of the great challenges in autonomous driving. We can suggest you to look at some research works like [this one](https://arxiv.org/pdf/1710.02410) or [this repo](https://github.com/JahodaPaul/autonomous-car-chase), where models trained in CARLA simulations have been implemented in real toy physical vehicles, similar to your approach. Hope it helps.

thank you

closed 2024-09-30 by MarcelPiNacy-CVC