carla-simulator/carla #7578: Carla UE4 Server crash

rowanabdelaleem, 2024-05-05:
CARLA version: 
0.9.15
Platform/OS: 
Ubuntu 22.04
Problem you have experienced: 
I was trying to train an RL agent using stable-baselines3 but this memory error occurs during the training and the server crash
[image]
Also during training this error appears: 
[image]

What you expected to happen: 
I expected to train without crashing the server
Steps to reproduce:
Other information (documentation you consulted, workarounds you tried):

Blyron, 2024-05-05:
We need more context. Which map are you using, Which gpu are you using, which NVIDIA Drivers are you using?

Blyron, 2024-05-05:
Before the crash it seems you have tried to close server without closing the client which creates issue.

rowanabdelaleem, 2024-05-05:
I`m using town 01  
GPU Type: NVIDIA GeForce RTX 4070 Ti
Driver Version: 550.54.15
and here is the environment I made:
class CarEnv(gym.Env):
    SHOW_CAM = SHOW_PREVIEW
    STEER_AMT = 1.0
    im_width = WIDTH
    im_height = HEIGHT
    front_camera = None
    CAMERA_POS_Z = 1.3 
    CAMERA_POS_X = 1.4
    PREFERRED_SPEED = 60
    SPEED_THRESHOLD = 2
    
    def __init__(self):
        super(CarEnv, self).__init__()
        
        self.action_space = spaces.Discrete(9)

        self.observation_space = spaces.Box(low=0.0, high=255.0,
                                            shape=(HEIGHT, WIDTH, N_CHANNELS), dtype=np.uint8)
        
        self.client = carla.Client("localhost", 2000)
        self.client.set_timeout(4.0)
        self.client.load_world('Town01')
        self.world = self.client.get_world()

        self.settings = self.world.get_settings()
        self.settings.no_rendering_mode = True
        #self.settings.synchronous_mode = False
        self.settings.fixed_delta_seconds = FIXED_DELTA_SECONDS
        self.world.apply_settings(self.settings)
        self.blueprint_library = self.world.get_blueprint_library()
        self.model_3 = self.blueprint_library.filter("model3")[0]
        
    def maintain_speed(self,s):
        
        if s >= self.PREFERRED_SPEED:
            return 0
        elif s < self.PREFERRED_SPEED - self.SPEED_THRESHOLD:
            return 0.7 # think of it as % of "full gas"
        else:
            return 0.3 # tweak this if the car is way over or under preferred speed 

        
    def reset(self, seed=None, options=None):
        self.collision_hist = []
        self.actor_list = []
        self.spawn_points = self.world.get_map().get_spawn_points()
        #print(type(self.spawn_points[2]))
        #print(type(random.choice(self.spawn_points)))
        
        self.vehicle = self.world.try_spawn_actor(self.model_3, random.choice(self.spawn_points))

    
        self.actor_list.append(self.vehicle)
        self.initial_location = self.vehicle.get_location()
        self.sem_cam = self.blueprint_library.find('sensor.camera.semantic_segmentation')
        self.sem_cam.set_attribute("image_size_x", f"{self.im_width}")
        self.sem_cam.set_attribute("image_size_y", f"{self.im_height}")
        self.sem_cam.set_attribute("fov", f"90")

        camera_init_trans = carla.Transform(carla.Location(z=self.CAMERA_POS_Z,x=self.CAMERA_POS_X))
        self.sensor = self.world.spawn_actor(self.sem_cam, camera_init_trans, attach_to=self.vehicle)
        self.actor_list.append(self.sensor)
        self.sensor.listen(lambda data: self.process_img(data))

        self.vehicle.apply_control(carla.VehicleControl(throttle=0.0, brake=0.0))
        time.sleep(2) #not detect a collision when the car spawns/falls from sky.
[... 83 lines cut ...]
        if self.episode_start + SECONDS_PER_EPISODE < time.time():
            done = True
            self.cleanup()
        
        observation = cam
        
        return observation.astype(np.uint8), reward, done, False, {}

    def cleanup(self):
        for sensor in self.world.get_actors().filter('*sensor*'):      
            sensor.destroy()
        for actor in self.world.get_actors().filter('*vehicle*'):
                    
            actor.destroy()
            
        cv2.destroyAllWindows()
        
    def process_img(self, image):
        image.convert(carla.ColorConverter.CityScapesPalette)
        i = np.array(image.raw_data)
        i = i.reshape((self.im_height, self.im_width, 4))[:, :, :3] # this is to ignore the 4th Alpha channel - up to 3
        self.front_camera = i/255.0

    def collision_data(self, event):
        self.collision_hist.append(event)

Blyron, 2024-05-06:
Yes, the RenderScreen used for CARLA server is only debug purpouses. Latest NVIDIA drivers are unstable for Vulkan. We recommend what UE recommend for UE4 which is 4XX

closed 2024-05-24 by Blyron