carla-simulator/carla #7579: make PythonAPI error [FAILED: src/CMakeFiles/xerces-c.dir/xercesc/util/regx/RegxParser.cpp.o ][20 errors generated. ninja: build stopped: subcommand failed.]

mohussein97, 2024-05-06:
Thanks for contributing to CARLA!

If you are reporting an issue, please use the following outline:

CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04.6
Problem you have experienced: I ran make python command and that is what I get [fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
[34/376] Building CXX object src/CMake...dir/xercesc/util/regx/RegxParser.cpp.o
FAILED: src/CMakeFiles/xerces-c.dir/xercesc/util/regx/RegxParser.cpp.o 
/home/mohammed/UnrealEngine-carla/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++ -DHAVE_CONFIG_H=1 -DXERCES_BUILDING_LIBRARY=1 -D_FILE_OFFSET_BITS=64 -D_THREAD_SAFE=1 -I/home/mohammed/carla/Build/xerces-c-3.2.4-source/build -I/home/mohammed/carla/Build/xerces-c-3.2.4-source/src -I/home/mohammed/carla/Build/xerces-c-3.2.4-source/build/src -std=c++14 -stdlib=libc++ -fPIC -w -I/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1 -L/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/lib/Linux/x86_64-unknown-linux-gnu -Wall -Wcast-align -Wcast-qual -Wctor-dtor-privacy -Wextra -Wformat=2 -Wimplicit-atomic-properties -Wmissing-declarations -Wno-long-long -Woverlength-strings -Woverloaded-virtual -Wredundant-decls -Wreorder -Wswitch-default -Wunused-variable -Wwrite-strings -Wno-variadic-macros -fstrict-aliasing -msse2 -O3 -DNDEBUG -MD -MT src/CMakeFiles/xerces-c.dir/xercesc/util/regx/RegxParser.cpp.o -MF src/CMakeFiles/xerces-c.dir/xercesc/util/regx/RegxParser.cpp.o.d -o src/CMakeFiles/xerces-c.dir/xercesc/util/regx/RegxParser.cpp.o -c /home/mohammed/carla/Build/xerces-c-3.2.4-source/src/xercesc/util/regx/RegxParser.cpp
In file included from /home/mohammed/carla/Build/xerces-c-3.2.4-source/src/xercesc/util/regx/RegxParser.cpp:25:
In file included from /home/mohammed/carla/Build/xerces-c-3.2.4-source/src/xercesc/util/regx/RegxParser.hpp:31:
In file included from /home/mohammed/carla/Build/xerces-c-3.2.4-source/src/xercesc/util/RefVectorOf.hpp:25:
In file included from /home/mohammed/carla/Build/xerces-c-3.2.4-source/src/xercesc/util/BaseRefVectorOf.hpp:25:
In file included from /home/mohammed/carla/Build/xerces-c-3.2.4-source/src/xercesc/util/ArrayIndexOutOfBoundsException.hpp:25:
In file included from /home/mohammed/carla/Build/xerces-c-3.2.4-source/src/xercesc/util/XMLException.hpp:25:
In file included from /home/mohammed/carla/Build/xerces-c-3.2.4-source/src/xercesc/util/XercesDefs.hpp:46:
In file included from /home/mohammed/carla/Build/xerces-c-3.2.4-source/build/src/xercesc/util/Xerces_autoconf_config.hpp:96:
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:152:8: error: no member named 'int8_t' in the global namespace
using::int8_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:153:8: error: no member named 'int16_t' in the global namespace
using::int16_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:154:8: error: no member named 'int32_t' in the global namespace
using::int32_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:155:8: error: no member named 'int64_t' in the global namespace
using::int64_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:157:8: error: no member named 'uint8_t' in the global namespace
using::uint8_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:158:8: error: no member named 'uint16_t' in the global namespace
using::uint16_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:159:8: error: no member named 'uint32_t' in the global namespace
using::uint32_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:160:8: error: no member named 'uint64_t' in the global namespace
using::uint64_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:162:8: error: no member named 'int_least8_t' in the global namespace
using::int_least8_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:163:8: error: no member named 'int_least16_t' in the global namespace
using::int_least16_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:164:8: error: no member named 'int_least32_t' in the global namespace
using::int_least32_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:165:8: error: no member named 'int_least64_t' in the global namespace
using::int_least64_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:167:8: error: no member named 'uint_least8_t' in the global namespace
using::uint_least8_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:168:8: error: no member named 'uint_least16_t' in the global namespace
using::uint_least16_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:169:8: error: no member named 'uint_least32_t' in the global namespace
using::uint_least32_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:170:8: error: no member named 'uint_least64_t' in the global namespace
using::uint_least64_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:172:8: error: no member named 'int_fast8_t' in the global namespace
using::int_fast8_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:173:8: error: no member named 'int_fast16_t' in the global namespace
using::int_fast16_t;
     ~~^
/home/mohammed/UnrealEngine-carla/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstdint:174:8: error: no member named 'int_fast32_t' in the global namespace
using::int_fast32_t;
     ~~^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
ninja: build stopped: subcommand failed.
make: *** [Util/BuildTools/Linux.mk:142: setup] Error 1
]
What you expected to happen: build success

Other information (documentation you consulted, workarounds you tried): official carla linux documentation, carla github, stack overflow, chat gpt-4 

[image]

michal1domanski, 2024-05-09:
I have the same issue. Does anybody has any solution for this one?

michal1domanski, 2024-05-09:
I have the same issue. Does anyone has a solution for this?

PatrickPromitzer, 2024-05-13:
Hi,
I had this error before, and fixed it.
You can try using
```
export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:/usr/bin/python3.8"
```
(I added it in my .bashrc)
You may have to change the python version in the export.

Another part I changed was adding
```
#include <stdint.h>
``` 
To the script that gives the error.
For my part, it was in "LibCarla/source/Carla/client/FileTransfer.h"
I believe the "RegxParser.cpp.o" gives the error, but you probably need to find if that it the root of it.

I am not sure if it fixes your problem, but I read that in an version update of a package the int and uint values changed (different name?) and this it raising an error.

xavisolesoft, 2024-05-17:
I suggest you to try instal:
```
sudo apt install libc6
```

And the run make agian.

michal1domanski, 2024-07-14:
Okay so I found out it was solved in #5209. It appears that this is completly not connected to the missing libraries. You have to switch name of the folder /usr/include/c++/v1 to something else (I changed it to v1.1) and it works.

DianweiCHEN, 2025-05-10:
> Hi, I had this error before, and fixed it. You can try using
>
> ```
> export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:/usr/bin/python3.8"
> ```
>
> (I added it in my .bashrc) You may have to change the python [...]

Thank you for the comment! I solved my problem!