carla-simulator/carla #7580: Ensure condition failed, can't launch Carla

flschnepf, 2024-05-06:
Hello everyone,

I am currently trying to install carla under ubuntu. I have followed the steps in the instructions and am now faced with the problem that I get error messages when starting unreal and carla. 
[image]

[carla_errors.txt](https://github.com/carla-simulator/carla/files/15219113/carla_errors.txt)

UE4 opens, but Carla won't start, both with the same error message.
I have installed the latest driver for the RTX 2080 ti and Vulkan also recognizes the graphics card according to "vkcube". 

Does anyone know what the problem is and how I can solve it?

CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04

Blyron, 2024-05-06:
This issue is known and Carla should start if you press play in editor

Blyron, 2024-05-06:
What happens when you make launch?
It gets stuck at 95%?

flschnepf, 2024-05-06:
Yes,

Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/florian/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182] 

/home/florian/carla/Util/BuildTools/BuildCarlaUE4.sh: line 205: 13389 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI} ${EDITOR_FLAGS}
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 137
`Ensure condition failed: VendorId != EGpuVendorId::Unknown [File:/home/florian/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp] [Line: 182] 

/home/florian/carla/Util/BuildTools/BuildCarlaUE4.sh: line 205: 13389 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI} ${EDITOR_FLAGS}
make: *** [Util/BuildTools/Linux.mk:8: launch] Error 137`

flschnepf, 2024-05-06:
[image]
It loads up to 95% and then the UE loading screen disappears. I do not stop the process.

Blyron, 2024-05-06:
That issue is related to Unreal Engine with vulkan. I suggest you to retry until it works and shaders are compiled or downgrade nvidia drivers and continue trying,

Blyron, 2024-05-06:
Please have patient. Those are warnings, not error messages. They are not important. Then, please wait 10-15 mins before pressing play, Unreal is still compiling when it starts, make launch, wait for 15 mins or until console says is not compiling shaders and try again

Blyron, 2024-05-17:
Are you happy to close this issue? 
Have you fixed your issue?

flschnepf, 2024-05-17:
Couldn't solve the problem. I think there is an issue witch my pc

Blyron, 2024-05-30:
We are gonna close this issue as more info is provided

closed 2024-05-30 by Blyron