carla-simulator/carla #7596: Wrong boundingbox size of Motorcycles, which is (0,0,0).

sgNicola, 2024-05-07:
CARLA version: Carla 0.9.14
Platform/OS: Ubuntu20.04
Problem you have experienced: Boundingbox extents for motorcycles are incorrectly set to (0,0,0), it might be due to a bug or misconfiguration in the simulator.  Issues with bounding box sizes can impact the evaluations' accuracy, such as IoU for object detection tasks.  
 
What you expected to happen: Fix the wrong bounding box dimensions.
Steps to reproduce: 
```
object_blueprint = blueprint_library.find('vehicle.mitsubishi.fusorosa')
sp1 = carla.Transform(carla.Location(x=-106.8, y=28.6, z=0.5),carla.Rotation(pitch=0, yaw=-90, roll=-20))
object = world.try_spawn_actor(object_blueprint, sp1)
bounding_box = object.bounding_box
print(bounding_box.extent.x, bounding_box.extent.y,bounding_box.extent.z)
```

Other information (documentation you consulted, workarounds you tried): 
[image]
[image]

sgNicola, 2024-05-07:
@Blyron, Could you help confirm the problem?

sgNicola, 2024-05-08:
Could you please verify the bug? @Axel1092 @corkyw10 @nsubiron @germanros1987 @bernatx @Blyron

MarcelPiNacy-CVC, 2024-05-10:
> Could you please verify the bug? @Axel1092 @corkyw10 @nsubiron @germanros1987 @bernatx @Blyron

Hi,

I will look into this.
Do not mass @ users too.

Marcel

PatrickPromitzer, 2024-05-13:
Hi,

I know that the bounding box values are not accurate, but for the actors you are using, it shouldn't be 0.0

```
vehicle.harley-davidson.low_rider
"size": {
        "x": 2.350175619125366,
        "y": 0.7662330269813538,
        "z": 1.6494941711425781
      },
"center": {
        "x": 0.0,
        "y": 7.629394360719743e-08,
        "z": 0.8246490359306335
      }
```

```
vehicle.kawasaki.ninja
"size": {
        "x": 2.043684244155884,
        "y": 0.7969123125076294,
        "z": 1.523191213607788
      },
"center": {
        "x": 0.0,
        "y": 7.629394360719743e-08,
        "z": 0.7688607573509216
      }
```

If you are using synchronize mode, you can try to add a world.tick() before getting the actor information.
(it could be that it is not updated)

If it still doesn't work, you could use the *.json from this issue for a workaround.
https://github.com/carla-simulator/carla/issues/7007

sgNicola, 2024-05-13:
> Hi,
>
> I know that the bounding box values are not accurate, but for the actors you are using, it shouldn't be 0.0
>
> ```
> vehicle.harley-davidson.low_rider
> "size": {
>         "x": [...]

Hi, I have tried add a world.tick() before getting the actor information, the values are still zero.
[image]

MarcelPiNacy-CVC, 2024-05-15:
Hi again,

I have tried to reproduce this. No luck so far though.
Are you running a packaged version of CARLA?
Also, are you able to upgrade to 0.9.15 if it comes to it?

Marcel

sgNicola, 2024-05-15:
Hi, I am running a packaged version (0.9.14) of Carla, I will try to upgrade to 0.9.15.

weiyangdaren, 2024-06-05:
Hi, have you resolved the issue? I encountered the same problem in version 0.9.15.

sgNicola, 2024-06-06:
Hi, not yet.

> Hi, have you resolved the issue? I encountered the same problem in version 0.9.15.